StanfordVL/GibsonEnv #7: ValueError: cannot reshape array of size 262144 into shape (64,64)

jakebruce, 2018-04-19:
Something about image resolutions appears to be broken. Running husky_navigate.yaml with resolutions other than 512 is causing a numpy reshape error for me:

```
killing <subprocess.Popen object at 0x7fe8c00baa90>
   File "./run", line 51, in <module>
   File "/usr/local/lib/python3.5/dist-packages/gym/core.py", line 104, in reset
   File "/home/jake/local-workspace/GibsonEnv/gibson/envs/husky_env.py", line 155, in _reset
   File "/home/jake/local-workspace/GibsonEnv/gibson/envs/env_modalities.py", line 334, in _reset
   File "/home/jake/local-workspace/GibsonEnv/gibson/envs/env_modalities.py", line 144, in _reset
   File "/home/jake/local-workspace/GibsonEnv/gibson/envs/env_modalities.py", line 457, in render_observations
   File "/home/jake/local-workspace/GibsonEnv/gibson/core/render/pcrender.py", line 496, in renderOffScreen
   File "/home/jake/local-workspace/GibsonEnv/gibson/core/render/pcrender.py", line 355, in render
 ValueError: cannot reshape array of size 262144 into shape (64,64)
```

I wasn't able to track down the root cause of this issue, although it only started happening after I started trying out different resolutions in the config file. Maybe some sort of caching problem?

fxia22, 2018-04-19:
You installed gibson from source right?

jakebruce, 2018-04-19:
That's right. Python 3.5.2, using build.sh.

fxia22, 2018-04-19:
Cool :+1:

closed 2018-04-19 by fxia22