StanfordVL/GibsonEnv #9: Very low fps

floodsung, 2018-04-30:
I run play_husky_camera.py on GTX 1070 but can only obtain an fps of 9, far less than what you reported on the paper. 
Do you know what is the problem?

fxia22, 2018-04-30:
@floodsung Thanks for point this out. You should be able to get slightly higher framerate with latest version of `master` branch now. In the paper, we used a smaller model (19% original size) to do framerate benchmarking and RL training, we will release those models soon too, stay tuned!

hzyjerry, 2018-05-01:
@floodsung a further note on low fps, displaying UI when running the environment causes speed drop too. If you set `display_ui: false` and `mode: headless` in examples/configs/play_husky_camera.yaml, the framerate should be higher.

closed 2018-05-03 by fxia22

chaundm, 2019-12-23:
Hi,

I have the same problem when using RTX 2070 but the fps is around 21 with the 256x256 resolution. I have tried to change in the config file but fps was still unchanged. 

I also tried with and without X server but the fps was around 21 for both cases. Is it normal for my GPU card or Could I am able to increase the fps more?