StanfordVL/GibsonEnv #16: Headless server deployment error

nsavinov, 2018-06-18:
Hi, I followed "Notes on deployment on a headless server" but I'm getting an error:
```Shell
root@feb123451b79:~/mount/gibson# DISPLAY=:1 vglrun python examples/demo/play_husky_nonviz.py
pybullet build time: Jun 18 2018 19:32:26
/root/mount/gibson/examples/demo/../configs/play/play_husky_nonviz.yaml
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
No protocol specified
[VGL] ERROR: Could not open display :0.
```
Output from /opt/websockify/run:
```Shell
WebSocket server settings:
  - Listen on :5901
  - Flash security policy server
  - Web server. Web root: /opt/noVNC
  - SSL/TLS support
  - proxying from :5901 to 'vncserver :1 -securitytypes otp -otp -noxstartup' (port 35923)
Starting 'vncserver :1 -securitytypes otp -otp -noxstartup'
xauth:  file /root/.Xauthority does not exist

Desktop 'TurboVNC: unix:1 ()' started on display unix:1

One-Time Password authentication enabled.  Generating initial OTP ...
Full control one-time password: 14299659
Run 'vncpasswd -o' from within the TurboVNC session or
    'vncpasswd -o -display unix:1' from within this shell
    to generate additional OTPs
Log file is /root/.vnc/unix:1.log
```
Log file contents:
```Shell
TurboVNC Server (Xvnc) 64-bit v2.1.2 (build 20170925)
Copyright (C) 1999-2017 The VirtualGL Project and many others (see README.txt)
Visit http://www.TurboVNC.org for more information on TurboVNC

18/06/2018 19:41:07 Using auth configuration file /etc/turbovncserver-security.conf
18/06/2018 19:41:07 Enabled authentication method 'otp'
18/06/2018 19:41:07 Advertising security type 'vncauth' to viewers
_XSERVTransmkdir: Mode of /tmp/.X11-unix should be set to 0777
_XSERVTransmkdir: this may cause subsequent errors
18/06/2018 19:41:07 Desktop name 'TurboVNC: unix:1 ()' (feb123451b79:1)
18/06/2018 19:41:07 Protocol versions supported: 3.3, 3.7, 3.8, 3.7t, 3.8t
18/06/2018 19:41:07 Listening for VNC connections on TCP port 5901
18/06/2018 19:41:07   Interface 127.0.0.1
18/06/2018 19:41:07 NOTICE: HTTP server disabled per system policy
18/06/2018 19:41:07 Framebuffer: BGRX 8/8/8/8
18/06/2018 19:41:07 Maximum clipboard transfer size: 1048576 bytes
18/06/2018 19:41:07 VNC extension running!
```
Could you help me resolve this problem?

fxia22, 2018-06-19:
Are you doing this inside a docker? If you want to run this inside a docker, you will need to use nvidia-docker, and forward `/tmp/.X11-unix` to inside the docker container. The host needs to have `/usr/lib/xorg/Xorg` running.

nsavinov, 2018-06-19:
Both things are fine:  I have ```/tmp/.X11-unix``` and ```nvidia-smi``` produces output. I have closely followed your instructions for headless deployment.

fxia22, 2018-08-10:
Hi @nsavinov, I remember you asked me to update you if we have progress on headless rendering. Now we have a branch `egl` that can completely run in headless mode now (no X server required), you can check that out and follow the config file `GibsonEnv/examples/configs/play/play_husky_camera.yaml` to do headless rendering.

closed 2018-08-10 by nsavinov