StanfordVL/GibsonEnv #23: Top-down view of Gibson

pbw-Berwin, 2018-07-28:
How can I get the top-down view of the whole Gibson map just like showed like this?
[image]

Thank you!

fxia22, 2018-08-01:
@pbw-Berwin We used to have that feature but removed that in the latest release. The reason is that pybullet is slow when rendering this view that has a large polygon count and this drags down rendering speed. I would suggest that you save the robot pose and visualize in some other software in a post-hoc manner. Or if you don't care that much about speed, you can use `env.render_map()` function to generate that view.

pbw-Berwin, 2018-08-01:
Thanks for your reply!

closed 2018-08-01 by pbw-Berwin

hzyjerry, 2018-08-10:
Hi @pbw-Berwin, we've recently added a functionality to visualize robot trajectory in post-hoc manner. See: https://github.com/StanfordVL/GibsonEnv/tree/master/gibson/data#visualization. You will need to install blender and save robot positions while doing navigation in Gibson.

The advantage of this method is that the visualization code automatically "slice" the model open for you, which bullet doesn't do. Say you're navigating on 2nd floor in a 3-storey building, this code will remove the top floor to reveal the trajectory.

Let me know if you have any questions.