StanfordVL/GibsonEnv #24: X server error

anoopsonar30, 2018-08-03:
I followed the instructions in your readme and was successfully able to setup and run the docker file. However I get this error 

**Invalid MIT-MAGIC-COOKIE-1 keyxhost:  unable to open display ":0.0"**

on running the **xhost +local:root** command

_Successfully built 62993fc08d1b
Successfully tagged gibson:latest
drparadox30@mae2:/media/drparadox30/Data/GibsonEnv$ sudo docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v '/media/drparadox30/Data/GibsonEnv/gibson/assets/dataset':/root/mount/gibson/gibson/assets/dataset gibson
[sudo] password for drparadox30: 
root@adb10486efed:~/mount/gibson# exit
exit
drparadox30@mae2:/media/drparadox30/Data/GibsonEnv$ export DISPLAY=:0.0
drparadox30@mae2:/media/drparadox30/Data/GibsonEnv$ xhost +local:root
Invalid MIT-MAGIC-COOKIE-1 keyxhost:  unable to open display ":0.0"_

What could be the problem?

fxia22, 2018-08-03:
Do you have physical access to your computer and do you have a display attached to the computer? When you run `nvidia-smi` on your host machine can you see `/usr/lib/xorg/Xorg` running?

anoopsonar30, 2018-08-09:
Yes, I have physical access to the computer and a display is attached. Yes, I am able to see `/usr/lib/xorg/Xorg` on running nvidia-smi on the host machine but I do not see it when I run it inside docker.

Host Machine - 
```
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 390.48                 Driver Version: 390.48                    |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  GeForce GTX 108...  Off  | 00000000:68:00.0  On |                  N/A |
| 23%   31C    P8    13W / 250W |    997MiB / 11175MiB |      1%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
|    0      1122      G   /usr/lib/xorg/Xorg                            40MiB |
|    0      1156      G   /usr/bin/gnome-shell                          50MiB |
|    0      1430      G   /usr/lib/xorg/Xorg                           406MiB |
|    0      1563      G   /usr/bin/gnome-shell                         324MiB |
|    0      8621      G   ...-token=3EE4B12AC2B7C121CEB81758F32B3724   171MiB |
+-----------------------------------------------------------------------------+
```

anoopsonar30, 2018-08-09:
I was using an Ubuntu 18.04LTS machine for the above commands so I tried using 16.04 instead. Now I'm able to run the xhost command but I get the following error on the last docker run command. 

```
(py35) drparadox30@mae2:~/GibsonEnv$ sudo docker run gibson
[sudo] password for drparadox30: 
Unable to find image 'gibson:latest' locally
docker: Error response from daemon: pull access denied for gibson, repository does not exist or may require 'docker login'.
See 'docker run --help'.
```

anoopsonar30, 2018-08-09:
Using the entire command 

`docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/GibsonEnv/gibson/assets/dataset:/root/mount/gibson/gibson/assets/dataset gibson`

results in the same error

anoopsonar30, 2018-08-09:
I also tried using Method 1 of creating my own docker image instead of building from source but encountered the following error - 

```
Step 21/27 : RUN  apt-get install -y vim wget unzip
 ---> Using cache
 ---> e18c9987dabe
Step 22/27 : RUN  apt-get install -y libzmq3-dev
 ---> Using cache
 ---> c2db78b9cdb1
Step 23/27 : ADD  . /root/mount/gibson
Error processing tar file(exit status 1): unexpected EOF
```
on executing the `docker build . -t gibson` command

Using your prebuilt docker image results in the same pull access denied error I mentioned before.

fxia22, 2018-08-09:
For creating your own docker image, you can remove the `*.tar.gz` files after you decompress it to get rid of the `Error processing tar file(exit status 1): unexpected EOF` error.

For the pull access denied error, try `sudo docker run xf1280/gibson:0.2`

anoopsonar30, 2018-08-09:
This doesn't work - `sudo docker run xf1280/gibson:0.2`
It results in the same error.

```
(py35) drparadox30@mae2:/media/drparadox30/New/sim$ sudo docker pull xf1280/gibson:0.2
0.2: Pulling from xf1280/gibson
Digest: sha256:f983a3ae4e6855a82c19cfeec2305488c392e95c0b99bbfd88ba91e35d76a116
Status: Image is up to date for xf1280/gibson:0.2
(py35) drparadox30@mae2:/media/drparadox30/New/sim$ sudo docker run xf1280:0.2
Unable to find image 'xf1280:0.2' locally
docker: Error response from daemon: pull access denied for xf1280, repository does not exist or may require 'docker login'.
See 'docker run --help'.
```
So I tried `sudo docker run xf1280/gibson:0.2` and it gives literally no output (no error)

Trying `sudo docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v /home/drparadox30/GibsonEnv/gibson/assets/dataset:/root/mount/gibson/gibson/assets/dataset xf1280/gibson:0.2` launches the docker environment correctly but I'm not sure if this is accessing the docker file I created or the one pulled.

closed 2018-08-10 by fxia22