StanfordVL/GibsonEnv #25: deployment on a headless server problem??????

zzzkk2009, 2018-08-07:
when i run 

_/opt/websockify/run 5901 --web=/opt/noVNC --wrap-mode=ignore -- vncserver :1 -securitytypes otp -otp -noxstartup_

this command,i got the infomation like this:

_WebSocket server settings:
  - Listen on :5901
  - Flash security policy server
  - Web server. Web root: /opt/noVNC
  - SSL/TLS support
  - proxying from :5901 to 'vncserver :1 -securitytypes otp -otp -noxstartup' (port 33951)
Starting 'vncserver :1 -securitytypes otp -otp -noxstartup'

Desktop 'TurboVNC: unix:1 ()' started on display unix:1

One-Time Password authentication enabled.  Generating initial OTP ...
Full control one-time password: 23411271
Run 'vncpasswd -o' from within the TurboVNC session or
    'vncpasswd -o -display unix:1' from within this shell
    to generate additional OTPs
Log file is /root/.vnc/unix:1.log_

then, how can i run 

_Run gibson with DISPLAY=:1 vglrun python <gibson example or training>_

this command? that is not interactive any more, it is listening the connection, and print the log info.

anyone can help ?

thks

zzzkk2009, 2018-08-07:
for this command 

_/opt/websockify/run 5901 --web=/opt/noVNC --wrap-mode=ignore -- vncserver :1 -securitytypes otp -otp -noxstartup_

when i add _--daemon_  flag 

the command window become interactive, but i can't find the password for vnc login!

and when i run  DISPLAY=:1 vglrun python  examples/demo/play_husky_nonviz.py
got the errors like this:

root@b35cb79bd062:~/mount/gibson# DISPLAY=:1 vglrun python  examples/demo/play_husky_nonviz.py
pybullet build time: Mar 20 2018 23:59:46
/root/mount/gibson/examples/demo/../configs/play/play_husky_nonviz.yaml
startThreads creating 1 threads.
starting thread 0
started thread 0
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
No protocol specified
[VGL] ERROR: Could not open display :0.

fxia22, 2018-08-07:
For your first question, you can open a `tmux` session and put `/opt/websockify/run 5901 --web=/opt/noVNC --wrap-mode=ignore -- vncserver :1 -securitytypes otp -otp -noxstartup` in a panel and run in background. You can run `DISPLAY=:1 vglrun python xxxx` in another panel in tmux or in the shell.

zzzkk2009, 2018-08-08:
@fxia22 thanks for your reply.

there is still some problem i not understand.
i'm not familiar about tmux.  
i'm used the command like below:
  1.  tmux  
  2.  docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix/X0:/tmp/.X11-unix/X0 -v /home/docker-map-folder/gibson/assets:/root/mount/gibson/gibson/assets -p 5901:5901 xf1280/gibson_server:0.2
  3. create a new pane ( ctrl + b,  ")
  4.in the first pane, i put /opt/websockify/run 5901 --web=/opt/noVNC --wrap-mode=ignore -- vncserver :1 -securitytypes otp -otp -noxstartup the shell command, and run it.
  5. then i change to the second pane, but this pane was not in the docker container, so  i run 
       docker attach f4443771ecc4    the command into the container.
  however ,these two pane are synchronizing show info.

  i don't understand how run _/opt/websockify/run 5901 --web=/opt/noVNC --wrap-mode=ignore -- vncserver :1 -securitytypes otp -otp -noxstartup_  this command in background.

   beg for your reply and answer, thanks again.

zzzkk2009, 2018-08-08:
i use nohup  command & solved this problem, but i got the same problem with issue https://github.com/StanfordVL/GibsonEnv/issues/16

closed 2018-08-08 by zzzkk2009