StanfordVL/GibsonEnv #26: env.reset() and env.step() not working

bradleyemi, 2018-08-08:
Just did a fresh install of Gibson from source. When I run the following code:

```
from gibson.envs.husky_env import HuskyNavigateEnv

env = HuskyNavigateEnv(gpu_count=1, config=<path to examples/configs/husky_navigate.yaml>)
env.reset()
```
or 

```
env.step(0)
```

I get a not implemented error. 

When I try doing env._step(0), I get:

```
~/svl/GibsonEnv/gibson/envs/husky_env.py in _rewards(self, action, debugmode)
     55     def _rewards(self, action=None, debugmode=False):
     56         a = action
---> 57         potential_old = self.potential
     58         self.potential = self.robot.calc_potential()
     59         progress = float(self.potential - potential_old)

AttributeError: 'HuskyNavigateEnv' object has no attribute 'potential'
```

Not quite sure what to do. Just upgraded OpenAI gym from 0.9.4 to 0.10.5. Perhaps this is the issue? Still shouldn't `env` have a `step` and `reset` method that work?

fxia22, 2018-08-10:
I can reproduce this issue, if upgrade gym from 0.9.4 to 0.10.5 there will be an NotImplementedError. @hzyjerry and I will look into this.

hzyjerry, 2018-08-10:
This is indeed caused by API breaking changes in gym 0.10.5.
Currently we require [gym version 0.9.4](https://github.com/StanfordVL/GibsonEnv/blob/master/setup.py#L26).

Will add support for the latest gym soon.

hzyjerry, 2018-08-10:
Hi @bradleyemi this is fixed and tested in [commit](a09fffda08a9c1560d17e65d14c89dcbe1421ed6). You can now pull from master branch and it should work.

fxia22, 2018-08-10:
I tested and it works. Thanks @hzyjerry!

closed 2018-08-10 by fxia22