StanfordVL/GibsonEnv #34: train_husky_navigate_ppo2 episodes to converge

tspthomas, 2018-09-04:
Hi, I was able to run the Docker version successfully.

When I run the script `train_husky_navigate_ppo2.py` it runs correctly but the robot barely moves. When I run the other script like `train_husky_gibson_flagrun_ppo1.py`, it runs much faster.

Is it expected? Do you have any numbers on how many epochs the agents start to converge or move  around?

I believe it can take very long to converge, so I'd just like to confirm. Let me know if I'm missing something.

Thanks in advance!

fxia22, 2018-09-04:
Yes, this is expected. The distinction between flagrun experiment and navigate experiment is that in flagrun experiment, the target is given while in navigate experiment the target is learned from signals of reward. So that would take much longer. In our experiment it takes 6*10e5 time steps, which is about (6*10e5 / 250 = ) 24000 episodes.

fxia22, 2018-09-04:
Yes, this is expected. The distinction between flagrun experiment and navigate experiment is that in flagrun experiment, the target is given while in navigate experiment the target is learned from signals of reward. So that would take much longer. In our experiment it takes 6e5 time steps, which is about (6e5 / 250 = ) 2400 episodes.

tspthomas, 2018-09-04:
Thanks for your reply @fxia22 !

A second question: during your experiments with husky trainings, did the agent ever got stuck in the training process? I mean, for about 200 iterations in the same place and not moving?

I noticed that running the default scripts (both ppo1 and ppo2) end up in this position. I'm not sure whether I should be using the master version or not. 

Do you have any thoughts on that?

Again, thanks for the help!

tspthomas, 2018-09-05:
Thanks a lot for your reply @fxia22 ! I was worried it was an issue in my setup and I've tried in different machines to confirm.

Looking forward to testing a fixed version!

Regards,
Thomas

fxia22, 2018-09-26:
We have fixed this issue in master branch, it turns out a data loading issue. Now you can run the example with the following commands:

```
mkdir gibson/utils/models
python examples/train/train_husky_navigate_ppo2.py 
python examples/train/enjoy_husky_navigate_ppo2.py  --reload_name gibson/utils/models/<epoch number>
```

You can also run the PPO1 example. Thanks for your patience.