StanfordVL/GibsonEnv #43: Gibson on remote server hangs when loading

glewis17, 2018-09-27:
When running on Deep Learning AMI (Ubuntu) Version 12.0 on a p2.xlarge AWS machine, gibson hangs at line 128:`self.socket_mist.connect("tcp://localhost:{}").format(self.port-1)` in file `gibson/core/render/pcrender.py`. Any tips on what may be wrong?

fxia22, 2018-09-27:
For AWS, the one possible setup is to use this [AMI](https://aws.amazon.com/marketplace/pp/B076K31M1S) and use the NGC image [here](https://github.com/StanfordVL/GibsonEnv/wiki#how-can-i-run-gibson-on-ngc)

fxia22, 2018-10-01:
Hi Gene, please see the updated set up guide in the readme. It should be the same as the steps I walked you through last week in my office. Let me know how it goes.

closed 2018-10-01 by fxia22

glewis17, 2018-10-03:
Confirmed working on a new machine, thanks!