StanfordVL/GibsonEnv #44: Gibson fails when disabling all UI components while rendering RGB image to the agent

glewis17, 2018-09-28:
Below error message obtained when running `examples/demo/play_husky_camera.py`

`examples/configs/play/play_husky_camera.py`:

`...
display_ui: false
show_diagnostics: false
ui_num: 0
ui_components: []
...
output: [nonviz_sensor, rgb_filled]
...`

[image]

hzyjerry, 2018-09-30:
Hi @glewis17 thanks for finding this issue. It is due to improper data loading in this configuration.

We have fixed this in master branch. Another thing is that by setting `display_ui: false`, you will not see Gibson's pygame-based UI, but will still see pybullet's native UI.

If you'd like to completely hide UI, you can set `mode: headless` as following:

```...
display_ui: false
show_diagnostics: false
ui_num: 0
ui_components: []
...
output: [nonviz_sensor, rgb_filled]
...
mode: headless
...```

closed 2018-10-01 by fxia22

JiaXuanWang, 2019-03-14:
Hi @hzyjerry I have modified `play_husky_camera.yaml` as you said. But it still doesn't work. When I run `python examples/demo/play_husky_camera.py`, it stops as follows:
[image]
I found it stopped at the line in `gibson/core/render/pcrender.py`:
`mist_msg = self.socket_mist.recv()`
Can you give me some advice to solve this problem?

mathildecaron31, 2019-10-22:
Hi @JessieWang-wjx , could you fix your problem ? I'm facing the same issue