StanfordVL/GibsonEnv #49: 2D floor map for planning

karkuspeter, 2018-10-05:
It would be useful to generate the 2D occupancy map of a floor, e.g. for path planning. The [navigation benchmark scenarios](@https://github.com/StanfordVL/GibsonEnv/tree/master/gibson/data#navigation-benchmark-scenarios) were presumable generated given such floor maps, however, i cannot find this feature in the public API. 

Thanks!

fxia22, 2018-10-12:
Here is a code snippet that does this:

You will need to install [meshcut](https://github.com/demansou/meshcut_python3) and comment out [a few lines](https://github.com/demansou/meshcut_python3/blob/master/meshcut.py#L50-L51). 

```python
import sys
import matplotlib.pyplot as plt
import meshcut
import numpy as np

def load_obj(fn):
	verts = []
	faces = []
	with open(fn) as f:
		for line in f:
			if line[:2] == 'v ':
				verts.append(list(map(float, line.strip().split()[1:4])))
			if line[:2] == 'f ':
				face = [int(item.split('/')[0]) for item in line.strip().split()[-3:]]
				faces.append(face)
	verts = np.array(verts)
	faces = np.array(faces) - 1
	return verts, faces

if __name__ == "__main__":	
	fn = sys.argv[1]
	verts, faces = load_obj(fn)
	z =  np.min(verts[:,-1]) + 0.5 # your robot height
	cross_section = meshcut.cross_section(verts, faces, plane_orig=(0, 0, z), plane_normal=(0, 0, 1))
	plt.figure()
	for item in cross_section:
		for i in range(len(item) - 1):
			plt.plot(item[i:i+2,0],item[i:i+2,1], 'b')
	plt.show()
```

fxia22, 2018-10-12:
Sample output: 
[image]

closed 2018-11-12 by fxia22

karkuspeter, 2018-11-13:
Thank you, this was very helpful