StanfordVL/GibsonEnv #54: An Issue about Environment on my Computer

Berk035, 2018-10-19:
Hello everyone, 

I am a novice ubuntu and github user. Nowadays, I am studying Master Degree in EEE at my university. So, I want to use GibsonEnv. for my final thesis. I installed packages and requirements according to the instructions. Then, I could work first example. Unfortunately, Other examples which consist camera and RGB don't work and are closed themselves. I've read some issues about cuda render or something else.

First example works as following;
[image]

Whereas other examples have problem.. And the output is given below;

`(py35) deepsrv@deepsrv-System-Product-Name:~/GibsonEnv$ python examples/demo/play_husky_camera.py
Error: cuda renderer is not loaded, rendering will not work
pybullet build time: Oct 17 2018 10:24:42
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
/home/deepsrv/GibsonEnv/examples/demo/../configs/play/play_husky_camera.yaml
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 780/PCIe/SSE2
GL_VERSION=3.2.0 NVIDIA 410.48
GL_SHADING_LANGUAGE_VERSION=1.50 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.2.0 NVIDIA 410.48
Vendor = NVIDIA Corporation
Renderer = GeForce GTX 780/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = NVIDIA Corporation
killing None
Error in sys.excepthook:
Traceback (most recent call last):
  File "/home/deepsrv/GibsonEnv/gibson/envs/env_modalities.py", line 642, in camera_multi_excepthook
    self.r_camera_mul.terminate()
AttributeError: 'NoneType' object has no attribute 'terminate'

Original exception was:
Traceback (most recent call last):
  File "examples/demo/play_husky_camera.py", line 17, in <module>
    env = HuskyNavigateEnv(config=args.config, gpu_idx = args.gpu)
  File "/home/deepsrv/GibsonEnv/gibson/envs/husky_env.py", line 40, in __init__
    self.robot_introduce(Husky(self.config, env=self))
  File "/home/deepsrv/GibsonEnv/gibson/envs/env_modalities.py", line 349, in robot_introduce
    self.setup_rendering_camera()
  File "/home/deepsrv/GibsonEnv/gibson/envs/env_modalities.py", line 375, in setup_rendering_camera
    self.setup_camera_multi()
  File "/home/deepsrv/GibsonEnv/gibson/envs/env_modalities.py", line 671, in setup_camera_multi
    self.r_camera_mul = subprocess.Popen(shlex.split(render_main), shell=False)
  File "/home/deepsrv/anaconda3/envs/py35/lib/python3.5/subprocess.py", line 676, in __init__
    restore_signals, start_new_session)
  File "/home/deepsrv/anaconda3/envs/py35/lib/python3.5/subprocess.py", line 1289, in _execute_child
    raise child_exception_type(errno_num, err_msg)
FileNotFoundError: [Errno 2] No such file or directory: './depth_render'
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/home/deepsrv/anaconda3/envs/py35/lib/python3.5/site-packages/gym/utils/closer.py", line 67, in close
    closeable.close()
  File "/home/deepsrv/GibsonEnv/gibson/envs/env_modalities.py", line 487, in _close
    self.r_camera_mul.terminate()
AttributeError: 'NoneType' object has no attribute 'terminate'
`

fxia22, 2018-10-19:
Looks like you are compiling from source, but it is not successfully compiled. Can you show the output of build script? (the output of `./build.sh build_local`)

Berk035, 2018-10-19:
Yes, I built it from source. I send the command which you said me and the output is attached below. 
[error_output.txt](https://github.com/StanfordVL/GibsonEnv/files/2495590/error_output.txt)

hzyjerry, 2018-10-19:
Hi @Berk035! Are you using ubuntu 18.04?

It looks like your GLM/GLX library is not building correctly, which has been [similarly reported by other developers on 18.04](https://git.immc.ucl.ac.be/robotran/mbsysc/issues/117).

Here is a quick fix ([original](https://git.immc.ucl.ac.be/robotran/mbsysc/commit/3feee824e1ae683f8361d5b378634a16922385fd)) by adding 1 line to `GibsonEnv/gibson/core/channels/CMakeLists.txt`

> add_definitions(-D GLM_ENABLE_EXPERIMENTAL)

Berk035, 2018-10-20:
> Hi @Berk035! Are you using ubuntu 18.04?
>
> It looks like your GLM/GLX library is not building correctly, which has been [similarly reported by other developers on [...]

Hi @hzyjerry, yes I am using Ubuntu 18.04. Thank you for your advice, I'll try it and give a feedback.

fxia22, 2018-10-31:
Looks like the compilation of depth_renderer is still not successful. Can you show the log of compilation?

Berk035, 2018-11-01:
@fxia22  Hi again,
Sorry for my late response. I viewed that the log file syslog and observed the last line as:
"**Nov  1 09:59:08 deepsrv-System-Product-Name gnome-shell[2498]: setup_framebuffers: assertion 'width > 0' failed**" 
Also, I observed the log of compilation from dmesg as fallows:
[dmesg_output.txt](https://github.com/StanfordVL/GibsonEnv/files/2537542/dmesg_output.txt)

fxia22, 2018-11-01:
@Berk035 sorry for the confusion, by log of compilation I mean the output of `./build.sh build_local`

Berk035, 2018-11-01:
@fxia22 I've complied and attached new output file below now: 
[output_2.txt](https://github.com/StanfordVL/GibsonEnv/files/2537908/output_2.txt)

Berk035, 2018-11-12:
Is there any solution for now? Did you find any opportunity for look up the problem? I want to develop my programme on GibsonEnv. If there is no solution for Ubuntu 18.04, I will try it on Ubuntu 16.

fxia22, 2018-11-12:
Sorry for the delay, looks like `depth_render` compilation was successful, however it is not in `GibsonEnv/gibson/core/channels/depth_render`, can you show `ls GibsonEnv/gibson/core/channels/depth_render`? Did you run `pip install -e .` after the compilation?

Berk035, 2018-11-13:
Hello again,
Yes I ran `pip install -e.` after installation and saw Gibson is installed succesfully. Also you can see files in directory below:
[image]

Thanks for attention.

hzyjerry, 2018-11-13:
@Berk035 hi, it looks like your `depth_render` is successfully compiled. 

If your Gibson rendering is blacked out, I think this could be possibly due to your CUDA version (it looks like you're using CUDA 10?) 

I recommend using CUDA 8 since it's well tested with Gibson. CUDA 7.5 is the version on my computer and it also works.

Berk035, 2018-11-13:
@hzyjerry hi, I suppose that I would downgrade my drivers and operating system for working. Thanks for advice.

hzyjerry, 2018-11-13:
I just realize that CUDA 8 and below are not supported on Ubuntu 18.04. I think [installing CUDA 9.0](https://gist.github.com/Mahedi-61/2a2f1579d4271717d421065168ce6a73) is worth trying if you'd like to avoid reinstalling the OS.

Berk035, 2018-11-14:
@hzyjerry Thank you for your respond sir, I will try somethings to figure out. Also, you predicted true about my cuda version which is CUDA 10 and it doesn't work properly. I reinstalled OS to another computer as Ubuntu 16.04 without any upgrades and I hope it will work this time. Thank you all for your concern again.

hzyjerry, 2018-11-15:
Sounds good! Let us know how it goes

Berk035, 2018-11-21:
@hzyjerry Hi again, I was able to run the demo with camera on Ubuntu 16.04 and CUDA 7.5
Now, I will develop my learning algorithm as next step but fps values are very low and process stucks sometimes. Is there any advice? Thank you for your helps. 
[image]

fxia22, 2018-11-21:
@Berk035 Are you using a GTX 780? It makes sense that you get such frame rate, because Gibson is designed for modern GPUs, like 9/10 series and needs a lot of computation power to run(each of our frames goes through a neural network). If you think your frame rate is too low, you can reduce the resolution to 128x128 and design your learning algorithm based on that.

Berk035, 2018-11-22:
@fxia22  I use the GTX1050 now. But, the other machine uses GTX780 double graphic cards. I tried the programme on both of them and changing some parameters for optimal work. I will change resolution for much FPS. Thank you for everything again.

closed 2018-11-22 by Berk035

zubair-irshad, 2019-12-23:
@Berk035 Did you solve the low frame rate issue? I am getting the same issue on my end

Berk035, 2019-12-24:
@zubair-irshad Yes, it is solved. Please check your graphics options and manage your timestep&frame_skip relations in config file.

zubair-irshad, 2019-12-24:
Awesome. I am curious to know, are there any optimum graphic options settings?