StanfordVL/GibsonEnv #56: Randomize initial robot position

glewis17, 2018-10-23:
Is there a way to randomize the initial robot position that respects the constraints of the environment?

When I enable random initial position, it currently seems to spawn the robot into walls, outside the environment, etc.

fxia22, 2018-10-24:
I think you can use the floorplan generated in #49 to inform your spawning. You can get a few cross sections at different heights of the robot and then make sure the footprint of the robot doesn't overlap with the obstacles.

glewis17, 2018-10-24:
Thanks Fei! I'm thinking of doing collision checking before starting an episode as a short-term hack. Is leveraging the floorplan for constrained randomization part of the development roadmap, or would it be worthwhile to develop a personal solution?

fxia22, 2018-10-24:
@glewis17 Leveraging the floorplan and collision check at the beginning of the episode are definitely in the development roadmap. If you are going to do something in this direction, a pull request will be much appreciated!