StanfordVL/GibsonEnv #61: camera image shows abnormally

slamjie, 2018-12-03:
when I run this command “roslaunch gibson-ros turtlebot_gmapping.launch” in ROS. The camera image shows like this.
[image]

but,when I just run Gibson without ros ,like "examples/demo/play_husky_camera.py" , the camera image is good.  
Is it wrong when I install Gibson?
How to deal with it?

hzyjerry, 2018-12-03:
Yes, the prefilled image and depth both look correct. Double checking with @fxia22.

fxia22, 2018-12-04:
The problem with rgb_filled is likely to be a wrong version of pytorch is exposed to python 2.7. You can check if you are using the correct version of pytorch.

slamjie, 2018-12-04:
> The problem with rgb_filled is likely to be a wrong version of pytorch is exposed to python 2.7. You can check if you are using the correct version of pytorch.

I install gibson from source in python2.7 for using gibson in ros.

slamjie, 2018-12-04:
> The problem with rgb_filled is likely to be a wrong version of pytorch is exposed to python 2.7. You can check if you are using the correct version of pytorch.

I install gibson from source in python2.7 for using gibson in ros. So, I download pytorch for python 2.7 in its site. its version is 0.4.1 . 

> The problem with rgb_filled is likely to be a wrong version of pytorch is exposed to python 2.7. You can check if you are using the correct version of pytorch.

slamjie, 2018-12-04:
> The problem with rgb_filled is likely to be a wrong version of pytorch is exposed to python 2.7. You can check if you are using the correct version of pytorch.

My pytorch version is 0.4.1. When I change the pytorch version to 0.3.1 , it  has no effect.
I want to use gibson with ROS, I follow the [](https://github.com/StanfordVL/GibsonEnv/tree/master/examples/ros/gibson-ros) but in this line `Install gibson from source _following installation guide_ in python2.7. ` , I cannot open the _following installation guide_ . So I do not know whether I install gibson successfully. 
When I run the demo from GibsonEnv, I will run `export PATH="/home/jaco/anaconda2/bin:$PATH"` firstly and it shows good . When I run demos with ROS in gibson-ros, I wiil run `export PYTHONPATH=/usr/lib/python2.7/dist-packages/:/opt/ros/kinetic/lib/python2.7/dist-packages:/home/jaco/anaconda2/envs/py27/lib/python2.7/site-packages:/home/jaco/GibsonEnv:$PYTHONPATH ` firstly and  the camera image specially rgb_filled shows badly.  Does it make `rgb_filled`  show badly?

slamjie, 2018-12-05:
> You installation seems to be correct.
> 
> Did you remove conda related items from `PATH`?
> 
> ```
> echo $PATH | grep -oP "[^:;]+" | grep conda	## Remove these paths from $PATH
> ```
> 
> For changing PYTHONPATH, don't add things before original PYTHONPATH, just added the following into an empty PYTHONPATH: `/usr/lib/python2.7/dist-packages/`, `/opt/ros/kinetic/lib/python2.7/dist-packages`(ros python libraries), `<anaconda installation root>/anaconda2/envs/py27/lib/python2.7/site-packages`(gibson dependencies) and `<gibson root>`. They need to be in PYTHONPATH in correct order.

Thanks for reply.
When I run gibson with ROS , I remove the conda `PATH` and run 
`$ echo $PATH | grep -oP "[^:;]+" | grep conda` 
it shows nothing in the terminal.
For the PYTHONPATH , when I open a new terminal and run` echo $PYTHONPATH` ,it shows nothing. Then, I add  `export PYTHONPATH=/usr/lib/python2.7/dist-packages/:/opt/ros/kinetic/lib/python2.7/dist-packages:/home/jaco/anaconda2/envs/py27/lib/python2.7/site-packages:/home/jaco/GibsonEnv:$PYTHONPATH `
[image]

These are the output in the terminal when I run gibson with ROS: 

jaco@jaco:~$ echo $PYTHONPATH

jaco@jaco:~$ source /opt/ros/kinetic/setup.bash 
jaco@jaco:~$ source catkin_ws/devel/setup.bash 
jaco@jaco:~$ echo $PATH | grep -oP "[^:;]+" | grep conda
jaco@jaco:~$ export PYTHONPATH=/usr/lib/python2.7/dist-packages/:/opt/ros/kinetic/lib/python2.7/dist-packages:/home/jaco/anaconda2/envs/py27/lib/python2.7/site-packages:/home/jaco/GibsonEnv:$PYTHONPATH 
jaco@jaco:~$ echo $PYTHONPATH/usr/lib/python2.7/dist-packages/:/opt/ros/kinetic/lib/python2.7/dist-packages:/home/jaco/anaconda2/envs/py27/lib/python2.7/site-packages:/home/jaco/GibsonEnv:/opt/ros/kinetic/lib/python2.7/dist-packages
jaco@jaco:~$ roslaunch gibson-ros turtlebot_gmapping.launch 
... logging to /home/jaco/.ros/log/b1e0a61e-f82f-11e8-b960-4cedfb67e5ad/roslaunch-jaco-6012.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://jaco:37093/

SUMMARY
========

PARAMETERS
 * /cmd_vel_mux/yaml_cfg_file: /opt/ros/kinetic/...
 * /depthimage_to_laserscan/output_frame_id: /camera_depth_frame
 * /depthimage_to_laserscan/range_min: 0.45
 * /depthimage_to_laserscan/scan_height: 2
 * /robot/name: turtlebot
 * /robot/type: turtlebot
 * /robot_description: <?xml version="1....
 * /robot_state_publisher/publish_frequency: 5.0
 * /rosdistro: kinetic
 * /rosversion: 1.12.14

NODES
  /
    cmd_vel_mux (nodelet/nodelet)
    depthimage_to_laserscan (depthimage_to_laserscan/depthimage_to_laserscan)
    gibson_clock (gibson-ros/simulation_clock.py)
    joint_state_publisher (joint_state_publisher/joint_state_publisher)
    mobile_base_nodelet_manager (nodelet/nodelet)
    robot_state_publisher (robot_state_publisher/robot_state_publisher)
    rviz (rviz/rviz)
    slam_gmapping (gmapping/slam_gmapping)
    turtlebot_gibson_sim (gibson-ros/turtlebot_rgbd.py)

auto-starting new master
process[master]: started with pid [6023]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to b1e0a61e-f82f-11e8-b960-4cedfb67e5ad
process[rosout-1]: started with pid [6036]
started core service [/rosout]
process[joint_state_publisher-2]: started with pid [6052]
process[turtlebot_gibson_sim-3]: started with pid [6054]
[... 51 lines cut ...]
GL_RENDERER=GeForce GTX 1080/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 415.18
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA
finish loading shaders
100%|█████████████████████████████████████████████| 1/1 [00:00<00:00,  5.24it/s]
100%|█████████████████████████████████████████| 190/190 [00:20<00:00,  9.05it/s]
{'model_id': 'space7', 'is_discrete': False, 'random': {'random_initial_pose': False, 'random_init_z_range': [-0.1, 0.1], 'random_target_pose': False, 'random_init_x_range': [-0.1, 0.1], 'random_init_y_range': [-0.1, 0.1], 'random_init_rot_range': [-0.1, 0.1]}, 'ui_num': 2, 'envname': 'TurtlebotNavigateEnv', 'target_orn': [0, 0, 1.57], 'speed': {'timestep': 0.00417, 'frameskip': 10}, 'initial_pos': [-14.3, 5, 1.2], 'verbose': False, 'fov': 1.57, 'display_ui': True, 'resolution': 256, 'use_filler': True, 'target_pos': [-14.3, 45.07, 0.5], 'mode': 'gui', 'output': ['nonviz_sensor', 'rgb_filled', 'depth'], 'ui_components': ['RGB_FILLED', 'DEPTH'], 'initial_orn': [0, 0, 4.71], 'show_diagnostics': False}
Episode: steps:0 score:0
Episode count: 0
render to ui
Laser Pose= 0.0114853 0.0928093 -1.52656
render to ui
render to ui
render to ui
render to ui
render to ui
render to ui
render to ui
render to ui
render to ui
render to ui
render to ui
render to ui

[image]

fxia22, 2018-12-05:
Thanks, can you also show the output of `source activate py27` and then `pip list | grep torch`?

slamjie, 2018-12-05:
> Thanks, can you also show the output of `source activate py27` and then `pip list | grep torch`?

```
jaco@jaco:~$ export PATH="/home/jaco/anaconda2/bin:$PATH"
jaco@jaco:~$ source activate py27
(py27) jaco@jaco:~$ pip list | grep torch
torch                              0.3.1                  
torchvision                        0.2.1                  
You are using pip version 10.0.1, however version 18.1 is available.
You should consider upgrading via the 'pip install --upgrade pip' command.
(py27) jaco@jaco:~$ 
```

fxia22, 2018-12-05:
I see, it might be `torchvision` version issue, if you downgrade to 0.2.0 it might solve the problem. Can you try it? Thanks

slamjie, 2018-12-05:
> I see, it might be `torchvision` version issue, if you downgrade to 0.2.0 it might solve the problem. Can you try it? Thanks

OK, Thanks.  I will try it. 
But I have a question . Why does the `rgb_filled` shows good in gibson without ROS but shows bad in gibson with ROS?

fxia22, 2018-12-05:
I am not sure about it either. The issue looks like a pytorch issue because `rgb_prefilled` looks fine (the gibson rendering part). With ros the environment is slightly changed and pytorch might have linked to different static libraries.

fxia22, 2018-12-05:
@slamjie I can reproduce your issue if I upgrade `torchvision` to `0.2.1`, I will look into this issue further.

slamjie, 2018-12-05:
> I am not sure about it either. The issue looks like a pytorch issue because `rgb_prefilled` looks fine (the gibson rendering part). With ros the environment is slightly changed and pytorch might have linked to different static libraries.

Thank you very much. 
I change the torchversion version to 0.2.0 and the `rgb_filled` shows good finally. 

[image]

that's cool !!  Thanks again.

closed 2018-12-07 by fxia22