StanfordVL/GibsonEnv #62: How can I generate the visual perception(rgb image) of the agent at any arbitrary postions and poses?

colincqian, 2018-12-10:
My aim is to export the visual perception(rgb image or depth image) of the agent for any given position and pose.

hzyjerry, 2018-12-11:
You can use `env.robot.reset_new_pose(pos, orn)` to set the robot to arbitrary position and orientation. Pos is xyz position and orn is wxyz quaternion.

Note that doing so may lead to collision with environment such as walls, or initializing robot outside the building. You may find #49 helpful.

slamjie, 2018-12-13:
> You can use `env.robot.reset_new_pose(pos, orn)` to set the robot to arbitrary position and orientation. Pos is xyz position and orn is wxyz quaternion.
> 
> Note that doing so may lead to collision with environment such as walls, or initializing robot outside the building. You may find #49 helpful.

How can I take the robot to somewhere when Gibson is running?

I write a ROS service with `env.robot.reset_new_pose(pos, orn)` . When I send a pose to it  by calling the service ,I want the robot  jump to there without moving.
But when I test this service, Gibson shutdown like this:
```
Fatal Python error: (pygame parachute) Segmentation Fault
[turtlebot_gibson_sim-2] process has died [pid 15935, exit code -6, cmd /home/jaco/catkin_ws/src/gibson_ros/turtlebot_rgbd.py __name:=turtlebot_gibson_sim __log:=/home/jaco/.ros/log/a57ed4da-fea7-11e8-9acc-4cedfb67e5ad/turtlebot_gibson_sim-2.log].
log file: /home/jaco/.ros/log/a57ed4da-fea7-11e8-9acc-4cedfb67e5ad/turtlebot_gibson_sim-2*.log
```
This is the handle of the ros service: 
```
def handle_set_pose(req):
    Point.x = req.pos_x
    Point.y = req.pos_y
    Point.z = req.pos_z
    Quaternion.x = req.ori_x
    Quaternion.y = req.ori_y
    Quaternion.z = req.ori_z
    Quaternion.w = req.ori_w
    env.robot.reset_new_pose(Point, Quaternion)
    return
```
I would appreciate if you give me some advises about possible mistakes that I might make.

colincqian, 2019-01-04:
@hzyjerry Could you please provide the proper form of the actions for
` obs, rew, env_done, info = env.step(action)`? I found the action list to be meaningless.