StanfordVL/GibsonEnv #63: depth render error when rendering semantics

stbook, 2018-12-11:
GibsonEnv works fine on my machine when the output is just RGB and depth.
When I tried to output Semantics, I got this error.
```
/home/reza/Datasets/GibsonEnv/gibson/assets/dataset/space7/semantic.obj: size of temp vertices 193957, vertex indices 1686 out vertices 1686
From ply loaded total of 4 vertices
Semantic.obj file was loaded with success.
Parsing /home/reza/Datasets/GibsonEnv/gibson/assets/dataset/space7/semantic.mtl file for material textures.
  1%|▎                                                    | 1/190 [00:02<06:35,  2.09s/it]Number of loaded materials: 1686
Texture file was loaded with success, total: 1686
Indexing VBO total groups: 1686
 16%|████████▍                                           | 31/190 [00:05<00:27,  5.83it/s]Finished indexing vertices v 473694 uvs 473694 normals 473694 semantics 473694
Semantics 
 19%|██████████▏                                         | 37/190 [00:06<00:26,  5.81it/s]*** 
Error in `./depth_render': corrupted double-linked list (not small): 0x00000000020b9750 ***
```
And the process got stuck at,
```
Semantic.obj file was loaded with success.
Parsing /home/reza/Datasets/GibsonEnv/gibson/assets/dataset/space7/semantic.mtl file for material textures.
Number of loaded materials: 1686
Texture file was loaded with success, total: 1686
filename: /home/reza/Datasets/GibsonEnv/gibson/assets/dataset/space7/mesh_z_up.obj
Episode: steps:0 score:0
Episode count: 0
```
I'm wondering if there is way to make the rendering work.

fxia22, 2019-01-16:
@hzyjerry any updates?

bradleyemi, 2019-02-21:
I'm seeing the same issue. @stbook @hzyjerry any progress on a solution or workaround?

stbook, 2019-02-22:
I am not able to fix it yet.

half-potato, 2019-04-06:
I found the issue. An off by 1 error [here](https://github.com/StanfordVL/GibsonEnv/blob/e8cdb37a67887fd765a02294b258db2e4fc417c4/gibson/core/channels/common/objloader.cpp#L154). It should read `std::vector<unsigned int> vertexFaces(temp_vertices.size()+1);`.

half-potato, 2019-04-07:
Wait nevermind that's a separate, though legitimate, issue.

half-potato, 2019-05-15:
This is a bug with cudaGetDevice in which a race condition causes it to be called at the same time as another instance. I have no idea how to fix this.

half-potato, 2019-05-17:
Weirdly, I was able to fix this issue by initializing zmq after cuda instead of the other way around.

closed 2019-09-17 by stbook