StanfordVL/GibsonEnv #64: Gibson Goggle

jhpenger, 2018-12-12:
Can you guys provide more instructions on how to use Gibson's Goggle?
I am trying to convert RGB images to Gibson style to train an image classifier for identifying chairs.

I am running Gibson headless in a docker container.
Thanks,

fxia22, 2018-12-27:
@jhpenger One example of using gibson goggle can be found in https://github.com/StanfordVL/GibsonEnv/blob/master/examples/ros/gibson-ros/goggle.py. It requires a kinect camera and ROS, as the goggle accepts RGBD images and adapts into to gibson style images.

jhpenger, 2018-12-29:
@fxia22 Can you provide some commentation for that file? I'm not using video feed from a camera, just a bunch of chair images to train an image classifier. Or can you provide a very easy example of converting normal jpg/png to gibson style?
Thanks,