StanfordVL/GibsonEnv #71: 512 resolution environment gives error

Exception4U, 2019-01-21:
(py35) XXXXX@XXXXXX:~/codes/GibsonEnv$ python examples/demo/play_turtlebot_camera.py 
pybullet build time: Jan 10 2019 16:01:23
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
/home/tushar/codes/GibsonEnv/examples/demo/../configs/play/play_turtlebot_camera.yaml
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX TITAN X/PCIe/SSE2
GL_VERSION=3.2.0 NVIDIA 396.54.09
GL_SHADING_LANGUAGE_VERSION=1.50 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.2.0 NVIDIA 396.54.09
Vendor = NVIDIA Corporation
Renderer = GeForce GTX TITAN X/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
ven = NVIDIA Corporation
Processing the data:
Total 1 scenes 0 train 1 test
Indexing
  0%|                                                                                                          | 0/1 [00:00<?, ?it/s]number of devices found 1
Loaded EGL 1.4 after reload.
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX TITAN X/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 396.54.09
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA
finish loading shaders
100%|██████████████████████████████████████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  2.69it/s]
 13%|████████████                                                                                   | 24/190 [00:01<00:23,  7.12it/s] 15%|██████████████▌                                                                                | 29/190 [00:02<00:15, 10.40it/s]100%|██████████████████████████████████████████████████████████████████████████████████████████████| 190/190 [00:10<00:00, 18.29it/s]
{'initial_pos': [-14.3, 5, 1.2], 'ui_num': 2, 'fov': 1.57, 'model_id': 'space7', 'target_orn': [0, 0, 1.57], 'verbose': False, 'display_ui': True, 'show_diagnostics': True, 'mode': 'gui', 'initial_orn': [0, 0, 4.71], 'speed': {'frameskip': 1, 'timestep': 0.01}, 'target_pos': [-14.3, 45.07, 0.5], 'is_discrete': True, 'envname': 'TurtlebotNavigateEnv', 'random': {'random_init_rot_range': [-0.1, 0.1], 'random_target_range': 0.1, 'random_init_z_range': [-0.1, 0.1], 'random_init_x_range': [-0.1, 0.1], 'random_init_y_range': [-0.1, 0.1], 'random_target_pose': False, 'random_initial_pose': False}, 'ui_components': ['RGB_FILLED', 'DEPTH'], 'resolution': 512, 'output': ['nonviz_sensor', 'rgb_filled', 'depth'], 'use_filler': True}
Episode: steps:0 score:0
Episode count: 0
THCudaCheck FAIL file=/pytorch/torch/lib/THC/generic/THCTensorCopy.c line=20 error=77 : an illegal memory access was encountered
killing <subprocess.Popen object at 0x7f3ad486a2b0>
   File "examples/demo/play_turtlebot_camera.py", line 16, in <module>
   File "/home/tushar/codes/GibsonEnv/gibson/utils/play.py", line 107, in play
   File "/home/tushar/codes/GibsonEnv/gibson/envs/env_modalities.py", line 391, in _reset
   File "/home/tushar/codes/GibsonEnv/gibson/envs/env_modalities.py", line 540, in render_observations
   File "/home/tushar/codes/GibsonEnv/gibson/core/render/pcrender.py", line 493, in renderOffScreen
   File "/home/tushar/codes/GibsonEnv/gibson/core/render/pcrender.py", line 437, in render
   File "/home/tushar/miniconda3/envs/py35/lib/python3.5/site-packages/torch/autograd/variable.py", line 298, in cuda
   File "/home/tushar/miniconda3/envs/py35/lib/python3.5/site-packages/torch/autograd/_functions/tensor.py", line 201, in forward
   File "/home/tushar/miniconda3/envs/py35/lib/python3.5/site-packages/torch/_utils.py", line 69, in _cuda
 RuntimeError: cuda runtime error (77) : an illegal memory access was encountered at /pytorch/torch/lib/THC/generic/THCTensorCopy.c:2
0
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed

Berk035, 2019-04-02:
Hello @Exception4U, 

I see that you run the environment from miniconda3 and your problem seems about 'cuda' when you try to run. Does it work properly with other resolutions? If your answer is 'yes', I don't have any idea for error. Otherwise, this problem may be about cuda.

half-potato, 2019-05-17:
get a better gpu