StanfordVL/GibsonEnv #74: View Synthesis

JiaXuanWang, 2019-03-08:
What should I do if I just want to do "View Synthesis" this section?

fxia22, 2019-03-08:
The view synthesis part is implemented in https://github.com/StanfordVL/GibsonEnv/blob/master/gibson/core/render/pcrender.py

You can modify this part of code and bypass the physics engine to render images directly.

JiaXuanWang, 2019-03-09:
Thanks for your reply.
But I got an error:
`Traceback (most recent call last):
  File "gibson/core/render/pcrender.py", line 567, in <module>
    data = d[v]
  File "/slstore/xuan/GibsonEnv/gibson/data/datasets.py", line 355, in __getitem__
    normal_imgs = [self.loader(item) for item in normal_img_paths]
  File "/slstore/xuan/GibsonEnv/gibson/data/datasets.py", line 355, in <listcomp>
    normal_imgs = [self.loader(item) for item in normal_img_paths]
  File "/slstore/xuan/GibsonEnv/gibson/data/datasets.py", line 37, in default_loader
    img = cv2.cvtColor(cv2.imread(path), cv2.COLOR_BGR2RGB)#.convert('RGB')
cv2.error: OpenCV(4.0.0) /io/opencv/modules/imgproc/src/color.cpp:181: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'`
Our dataset does not have normal images, right?

charliememory, 2019-03-29:
Does the current code support to render the panoramic view? I see the pano flag is set to false in https://github.com/StanfordVL/GibsonEnv/blob/8e9e23c2dcc44bb5442a25d12928103f05b51d8f/gibson/core/render/pcrender.py#L370
And the depth pano render code is commented out https://github.com/StanfordVL/GibsonEnv/blob/8e9e23c2dcc44bb5442a25d12928103f05b51d8f/gibson/core/channels/depth_render/render.cpp#L920

cazhang, 2020-01-14:
@charliememory  any luck of using pano mode? I'd like to use the feature if possible.

charliememory, 2020-01-14:
@cazhang I didn't follow the pano mode. It seems this feature was not supported at that moment. I am also interested in this feature.

Changyangli, 2020-02-05:
I tried to use the pano mode but failed... What I am doing now is to take pictures from various views at the same location to build a skybox, and then stitch them together. Like this
[image]