StanfordVL/GibsonEnv #75: Do we plan to support lidar in Gibson?

susu3621, 2019-03-08:
We are investigating the possibility to use Gibson to replace Gazebo to simulate robotics.
But seems there is no lidar support in Gibson.
Do we have a plan to support lidar in Gibson? Or am I missed the lidar agent implementation?

fxia22, 2019-03-08:
@susu3621 It is implemented in a newer version of Gibson that is not released yet. Stay tuned :)

susu3621, 2019-03-09:
Great! Waiting for it.

botforge, 2019-08-01:
Hi, is this version of Gibson on any of the other branches or on a fork?

fxia22, 2019-08-01:
@botforge you can find it here: https://github.com/StanfordVL/GibsonEnvV2

The code to support scan is in these lines: https://github.com/fxia22/gibsonv2/blob/master/gibson2/envs/locomotor_env.py#L266-L302

botforge, 2019-08-01:
@fxia22 I think the second repository is private under your account. However, GibsonEnvV2 is not, is the code to support the latter is here?
https://github.com/StanfordVL/GibsonEnvV2/blob/d880d9b3de87a9867dabed2a2f2321fe47456aa4/gibson2/envs/locomotor_env.py#L266

fxia22, 2019-08-01:
@botforge yes, that's what I mean.