StanfordVL/GibsonEnv #77: Camera Intrinsic Matrix

stbook, 2019-03-29:
I know that GibsonEnv use NN to do the rendering. But is there a way to get to know the intrinsic matrix of the camera? Because I think the input parameter 'fov' doesn't affect the output RGB image here.

closed 2019-03-29 by stbook

placeforyiming, 2019-06-18:
> I know that GibsonEnv use NN to do the rendering. But is there a way to get to know the intrinsic matrix of the camera? Because I think the input parameter 'fov' doesn't affect the output RGB image here.

Hi,

Did you get that?

fxia22, 2019-06-19:
@placeforyiming It is a pin-hole camera model with a certain fov. So there is no skew, optical center is the image center and focal length can be calculated from field of view.

NaveenSehgal, 2019-07-23:
> @placeforyiming It is a pin-hole camera model with a certain fov. So there is no skew, optical center is the image center and focal length can be calculated from field of view.

How exactly is the focal length calculated from field of view? Don't you need sensor size to do this?

stbook, 2019-11-07:
> > I know that GibsonEnv use NN to do the rendering. But is there a way to get to know the intrinsic matrix of the camera? Because I think the input parameter 'fov' doesn't affect the output RGB image here.
> 
> Hi,
> 
> Did you get that?

If the resolution of the observation is (256, 256), then the focal length is 128.