StanfordVL/GibsonEnv #78: web_ui not working

imankgoyal, 2019-04-01:
Hi, 
I am having trouble running the web_ui mode on a headless server. I would very much appreciate any help in this regard. I am listing the steps here so as to make sure I am not missing anything.
 
- I changed the configs for play_husky_nonviz.yaml 
- I ran the gibson/utils/web_ui.py 
- In another screen I ran examples/demo/play_husky_nonviz.py

On connecting to the port (5001) on the server, I am seeing a webpage with 2 black boxes. I think the code for the environment is running fine and the issue is related to rendering. I have attached screenshots for illustration. 

Changes made to the config file:
[image]

Running gibson/utils/web_ui.py:
[image]

Running examples/demo/play_husky_nonviz.py: 
[image]

Rendered Webpage:
[image]

Also, the black boxes apprear only when I run examples/demo/play_husky_nonviz.py. I think it is expected that the page renders something when the video streams starts. 

Please let me know if some more information is required.
Thanks!

Berk035, 2019-04-02:
Hi @imankgoyal, 

This example does not open the user interface camera box as you can see from its name "play_husky__nonviz_.py". 

You can open `python examples/demo/play_husky_camera.py` from Gibson examples to see web_ui mode. Also, your streaming way is correct.

imankgoyal, 2019-04-02:
Hi @Berk035 ,

Thank you so much for the reply. I didn't notice that the web_ui mode won't display the entire agent as in the case of gui, but only display its camera outputs. Also, in the web_ui mode, can we control the agent with AWSD or is it only possible to do so in the gui mode?

As you suggested, I ran ``python examples/demo/play_husky_camera.py`` in the web_ui mode, but the environment now seems to be stuck (attached is the stdout). Essentially, I get the ``Unable to initialize EGL`` error message and then the console is struck.
[image]

I would very much appreciate any help in this regard. Please let me know if any more info is required.
Thanks!