StanfordVL/GibsonEnv #81: About Enjoy_Husky_Gibson_Flagrun_ppo1

Berk035, 2019-04-12:
Hello everyone,

I am studying on flagrun with ppo1 algorithm nowadays. Unfortunately, I have a problem while I am trying to run enjoy_husky_gibson_flagrun_ppo1.py.
The error is:
`TypeError: __init__() missing 2 required positional arguments: 'hid_size' and 'num_hid_layers'`
I realized that, model which consists information of size and layers is lost. How can I set these parameters? 

A second question: I see a line in the same file  as `reload_name = '/home/berk/GibsonEnv/gibson/examples/train/models/flagrun_RGBD2_50.model'`. There is not any file in this location as '.model' . How can I import the model to the code and create .model file?

Thank you in advance.

Berk035, 2019-04-21:
Up

fxia22, 2019-04-23:
Did you use `train_husky_gibson_flagrun_ppo1.py` script to train the model? You need to run that script first. Also `hid_size` and `num_hid_layers` can be recovered from that script as well.

Berk035, 2019-04-25:
Yes I did, but I haven't realized that the file and some trials are terminated due to size of resolution. I will try again with slight changes and inform you. 
Thank you

Berk035, 2019-04-29:
Hello again,
I tried to apply training the husky but I am still at same point. I couldn't the create model file. When I tried to run `train_husky_gibson_flagrun_ppo1.py`, I got the error which is given followings:

[image]
[image]

I couldn't see the problem in `shuffle()`. How can I handle with this error? 
Thank you.

Berk035, 2019-10-31:
Different issue continues on flagrun_ppo1 with:
`Connot feed value of shape (64,128,4) for Tensor 'pi/ob:0', which has shape '(?,128,128,4)`
I've changed all parameters but it doesn't match. Thanks for your helping.

Sincerely,
Berk

Berk035, 2019-11-22:
I fixed problem. It causes due to 'ob' variable. It is also discussed in issue of #55.

closed 2019-11-22 by Berk035