StanfordVL/GibsonEnv #82: PyBullet initialization

anoopsonar30, 2019-04-15:
I'm trying to implement quadrotor initialization in multiple environments using the waypoints data from the csv files in your database. It basically works by reading each line from the CSV file, copying it to a yaml config file and then using that config file to initialize the environment and then run the simulation(done by the precompute cost function which simply resets the each iteration on the same environment).

However after the first iteration (although the second config is loaded successfully) I get the following pybullet error - 

> Only one local in-process GUI/GUI_SERVER connection allowed. Use DIRECT connection mode or start a separate GUI physics server (ExampleBrowser, App_SharedMemoryPhysics_GUI, App_SharedMemoryPhysics_VR) and connect over SHARED_MEMORY, UDP or TCP instead.

This happens at the following line in the code below - 
`env = DroneNavigateEnv(config = args.config)` 

Since the Gibson environment is acting like a wrapper around the engine, I don't know how to directly access the underlying PyBullet engine. How do I fix the above error? Morever, how can I run Gibson in DIRECT mode? i.e. run simulations faster without the GUI? 

```
cnfgfile = os.path.join(os.path.dirname(os.path.realpath(__file__)), '.','config_pacbayes.yaml')
    costs_precomputed = np.zeros((numSimulations, L))
    envNum = 0

    csvfile = open('gibson_full.csv')
    reader = csv.reader(csvfile, delimiter=',')
    for row in reader:
        if ((envNum % 10) == 0):
            print(envNum,"out of ",numSimulations)

        model_id  = row[1]
        x = float(row[5])
        y = float(row[6])
        z = float(row[7])
        angle = float(row[8])

        with open(cnfgfile) as f:
            config_file = yaml.load(f)

        config_file['model_id'] = model_id
        config_file['initial_pos'] = [x, y, z]
        config_file['initial_orn'] = [0, 0, angle]

        print(config_file)
        with open(cnfgfile, 'w') as f:
            yaml.dump(config_file, f, default_flow_style=False)

        parser = argparse.ArgumentParser()
        parser.add_argument('--config', type=str, default=cnfgfile)
        args = parser.parse_args()

        env = DroneNavigateEnv(config = args.config)
        print(env.config)
        costs_precomputed[envNum] = precompute_environment_costs(K, L, env, numRays, T_horizon)
        envNum = envNum + 1
```

anoopsonar30, 2019-04-15:
Adding the line 'pybullet.connect(pybullet.DIRECT)' before 'env = DroneNavigateEnv(config = args.config)' does not fix the error even although the pybullet GUI screen gets deactivated and the simulations are running much faster.

anoopsonar30, 2019-04-15:
After editing my yaml config file to change display_ui to false, and changing the mode from gui to headless, the code seems to be working. However if run in GUI mode, the error still persists. How do I fix that? I need that for debugging purposes!

zubair-irshad, 2019-12-27:
@anoopsonar30 Did you find a work around this issue?

anoopsonar30, 2019-12-28:
Nope. Also I think they moved support for this repo to the GibsonEnvV2. I'm currently experimenting with Habitat instead.

zubair-irshad, 2019-12-30:
Thanks @anoopsonar30 
Any chance you are getting the following error while running env = DroneNavigateEnv(config = args.config) within a loop (headless mode)

**terminate called after throwing an instance of 'zmq::error_t'
  what():  Address already in use**

It seems like re-initializing environment within a loop allocates same memory address and hence it throws an error.