StanfordVL/GibsonEnv #89: Problems running train_husky_navigate_ppo2.py

jingxixu, 2019-05-24:
When I try to run `python examples/train/train_husky_navigate_ppo2.py`, it gives me the following error

```
pybullet build time: May 18 2019 17:58:05
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
2019-05-24 14:43:26.019178: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.1 instructions, but these are available on your machine and could speed up CPU computations.
2019-05-24 14:43:26.019197: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use SSE4.2 instructions, but these are available on your machine and could speed up CPU computations.
2019-05-24 14:43:26.019259: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX instructions, but these are available on your machine and could speed up CPU computations.
2019-05-24 14:43:26.019266: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use AVX2 instructions, but these are available on your machine and could speed up CPU computations.
2019-05-24 14:43:26.019272: W tensorflow/core/platform/cpu_feature_guard.cc:45] The TensorFlow library wasn't compiled to use FMA instructions, but these are available on your machine and could speed up CPU computations.
Logging to /tmp/openai-2019-05-24-14-43-26-020502
/home/jxu/gibson_ws/GibsonEnv/examples/train/../configs/husky_navigate_rgb_train.yaml
/home/jxu/gibson_ws/GibsonEnv/gibson/envs/env_bases.py:66: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  config_data = yaml.load(f)
Processing the data:
Total 1 scenes 0 train 1 test
Indexing
  0%|                                                                          | 0/1 [00:00<?, ?it/s]number of devices found 1
Loaded EGL 1.5 after reload.
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=GeForce GTX 1080/PCIe/SSE2
GL_VERSION=4.6.0 NVIDIA 418.67
GL_SHADING_LANGUAGE_VERSION=4.60 NVIDIA
finish loading shaders
100%|██████████████████████████████████████████████████████████████████| 1/1 [00:00<00:00,  6.12it/s]
 15%|█████████▎                                                     | 28/190 [00:03<00:21,  7.58it/s]terminate called after throwing an instance of 'zmq::error_t'
  what():  Address already in use
100%|██████████████████████████████████████████████████████████████| 190/190 [00:18<00:00, 10.22it/s]
/home/jxu/gibson_ws/GibsonEnv/gibson/core/render/pcrender.py:204: UserWarning: volatile was removed and now has no effect. Use `with torch.no_grad():` instead.
  self.imgv = Variable(torch.zeros(1, 3 , self.showsz, self.showsz), volatile = True).cuda()
/home/jxu/gibson_ws/GibsonEnv/gibson/core/render/pcrender.py:205: UserWarning: volatile was removed and now has no effect. Use `with torch.no_grad():` instead.
  self.maskv = Variable(torch.zeros(1,2, self.showsz, self.showsz), volatile = True).cuda()
/home/jxu/gibson_ws/GibsonEnv/gibson/utils
(1, 128, 128, 4)
(1, 23)
Episode: steps:0 score:0
Episode count: 0
killing <subprocess.Popen object at 0x7f308888e320>
   File "examples/train/train_husky_navigate_ppo2.py", line 112, in <module>
   File "examples/train/train_husky_navigate_ppo2.py", line 98, in main
   File "examples/train/train_husky_navigate_ppo2.py", line 71, in train
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/utils/ppo2.py", line 234, in learn
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/utils/ppo2.py", line 98, in __init__
   File "/home/jxu/anaconda3/envs/py35/lib/python3.5/site-packages/gym/core.py", line 104, in reset
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/utils/monitor.py", line 56, in _reset
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/envs/env_modalities.py", line 384, in _reset
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/envs/env_modalities.py", line 166, in _reset
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/envs/env_modalities.py", line 529, in render_observations
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/core/render/pcrender.py", line 493, in renderOffScreen
   File "/home/jxu/gibson_ws/GibsonEnv/gibson/core/render/pcrender.py", line 379, in render
 ValueError: cannot reshape array of size 65536 into shape (128,128)
```

I am able the run the other two examples:
```
python examples/demo/play_husky_nonviz.py ### Use ASWD keys on your keyboard to control a car to navigate around Gates building
python examples/demo/play_husky_camera.py ### Use ASWD keys on your keyboard to control a car to navigate around Gates building, while RGB and depth camera outputs are also shown.
```

Can someone help? Thank you!

Berk035, 2019-05-29:
Hello dear @jingxixu ,
I am very familiar to this issue due to working with flagrun and ppo1 example. You should debug your script and detect the size of tensors. It may be caused from configuration file arguments or inside of code. If I face with this problem again, I will introduce you with the exact point. Have a nice day.

jingxixu, 2019-05-30:
@Berk035 Could you please explain in details? Is it a python package version issue? Why is it working for others if there is a bug? Many thanks!

Berk035, 2019-05-30:
> @Berk035 Could you please explain in details? Is it a python package version issue? Why is it working for others if there is a bug? Many thanks!

I said that there is no bug, you should 'debug' your script. It's related with size of tensors, you can observe what is the wrong with debugging.

jingxixu, 2019-05-31:
I am sorry @Berk035, but I have no script. I am using their script `train_husky_navigate_ppo2.py` which is provided as an example. Are you saying this script can be problematic. Thanks!

jingxixu, 2019-06-09:
This problem is solved as indicated in [#7](https://github.com/StanfordVL/GibsonEnv/issues/7).

closed 2019-06-09 by jingxixu