StanfordVL/GibsonEnv #90: 'HuskyNavigateEnv' object has no attribute 'UI'

Yuantian013, 2019-05-29:
(base) weipan@TUD278688:~$ docker run --runtime=nvidia --rm nvidia/cuda nvidia-smi
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.39/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.
(base) weipan@TUD278688:~$ sudo docker run --runtime=nvidia --rm nvidia/cuda nvidia-smi
[sudo] password for weipan: 
Wed May 29 12:39:50 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.67       Driver Version: 418.67       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  TITAN V             On   | 00000000:65:00.0  On |                  N/A |
| 29%   43C    P0    37W / 250W |    396MiB / 12028MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
(base) weipan@TUD278688:~$ docker pull xf1280/gibson:0.3.1
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.39/images/create?fromImage=xf1280%2Fgibson&tag=0.3.1: dial unix /var/run/docker.sock: connect: permission denied
(base) weipan@TUD278688:~$ sudo docker pull xf1280/gibson:0.3.1
0.3.1: Pulling from xf1280/gibson
Digest: sha256:3f91fca5a46825e87fe4db1d7b6a3ad91d896d2e0bd73daafc81f0059094617c
Status: Image is up to date for xf1280/gibson:0.3.1
(base) weipan@TUD278688:~$ xhost +local:root
non-network local connections being added to access control list
(base) weipan@TUD278688:~$ xhost ls
xhost:  bad hostname "ls"
(base) weipan@TUD278688:~$ xhost
access control enabled, only authorized clients can connect
LOCAL:
SI:localuser:weipan
(base) weipan@TUD278688:~$ sudo docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v LOCAL:/root/mount/gibson/gibson/assets/dataset xf1280/gibson:0.3.1
root@b63b6a96e6d2:~/mount/gibson# python examples/demo/play_husky_nonviz.py
pybullet build time: Sep 27 2018 00:17:23
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
/root/mount/gibson/examples/demo/../configs/play/play_husky_nonviz.yaml
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=TITAN V/PCIe/SSE2
GL_VERSION=3.2.0 NVIDIA 418.67
GL_SHADING_LANGUAGE_VERSION=1.50 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.2.0 NVIDIA 418.67
Vendor = NVIDIA Corporation
Renderer = TITAN V/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
Traceback (most recent call last):
  File "examples/demo/play_husky_nonviz.py", line 14, in <module>
    env = HuskyNavigateEnv(config = args.config)
  File "/root/mount/gibson/gibson/envs/husky_env.py", line 38, in __init__
    tracking_camera=tracking_camera)
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 333, in __init__
    self.model_path = get_model_path(self.model_id)
  File "/root/mount/gibson/gibson/data/datasets.py", line 52, in get_model_path
    assert (model_id in os.listdir(data_path)) or model_id == 'stadium', "Model {} does not exist".format(model_id)
AssertionError: Model space7 does not exist
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/miniconda/envs/py35/lib/python3.5/site-packages/gym/utils/closer.py", line 67, in close
    closeable.close()
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 498, in _close
    self.UI._close()
AttributeError: 'HuskyNavigateEnv' object has no attribute 'UI'
Exception ignored in: <bound method Env.__del__ of <gibson.envs.husky_env.HuskyNavigateEnv object at 0x7f3b37df90b8>>
Traceback (most recent call last):
  File "/miniconda/envs/py35/lib/python3.5/site-packages/gym/core.py", line 203, in __del__
    self.close()
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 498, in _close
    self.UI._close()
AttributeError: 'HuskyNavigateEnv' object has no attribute 'UI'
root@b63b6a96e6d2:~/mount/gibson#

Yuantian013, 2019-05-29:
(base) weipan@TUD278688:~$ docker run --runtime=nvidia --rm nvidia/cuda nvidia-smi
docker: Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.39/containers/create: dial unix /var/run/docker.sock: connect: permission denied.
See 'docker run --help'.
(base) weipan@TUD278688:~$ sudo docker run --runtime=nvidia --rm nvidia/cuda nvidia-smi
[sudo] password for weipan: 
Wed May 29 12:39:50 2019       
+-----------------------------------------------------------------------------+
| NVIDIA-SMI 418.67       Driver Version: 418.67       CUDA Version: 10.1     |
|-------------------------------+----------------------+----------------------+
| GPU  Name        Persistence-M| Bus-Id        Disp.A | Volatile Uncorr. ECC |
| Fan  Temp  Perf  Pwr:Usage/Cap|         Memory-Usage | GPU-Util  Compute M. |
|===============================+======================+======================|
|   0  TITAN V             On   | 00000000:65:00.0  On |                  N/A |
| 29%   43C    P0    37W / 250W |    396MiB / 12028MiB |      0%      Default |
+-------------------------------+----------------------+----------------------+
                                                                               
+-----------------------------------------------------------------------------+
| Processes:                                                       GPU Memory |
|  GPU       PID   Type   Process name                             Usage      |
|=============================================================================|
+-----------------------------------------------------------------------------+
(base) weipan@TUD278688:~$ docker pull xf1280/gibson:0.3.1
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Post http://%2Fvar%2Frun%2Fdocker.sock/v1.39/images/create?fromImage=xf1280%2Fgibson&tag=0.3.1: dial unix /var/run/docker.sock: connect: permission denied
(base) weipan@TUD278688:~$ sudo docker pull xf1280/gibson:0.3.1
0.3.1: Pulling from xf1280/gibson
Digest: sha256:3f91fca5a46825e87fe4db1d7b6a3ad91d896d2e0bd73daafc81f0059094617c
Status: Image is up to date for xf1280/gibson:0.3.1
(base) weipan@TUD278688:~$ xhost +local:root
non-network local connections being added to access control list
(base) weipan@TUD278688:~$ xhost ls
xhost:  bad hostname "ls"
(base) weipan@TUD278688:~$ xhost
access control enabled, only authorized clients can connect
LOCAL:
SI:localuser:weipan
(base) weipan@TUD278688:~$ sudo docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v LOCAL:/root/mount/gibson/gibson/assets/dataset xf1280/gibson:0.3.1
root@b63b6a96e6d2:~/mount/gibson# python examples/demo/play_husky_nonviz.py
pybullet build time: Sep 27 2018 00:17:23
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
/root/mount/gibson/examples/demo/../configs/play/play_husky_nonviz.yaml
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=TITAN V/PCIe/SSE2
GL_VERSION=3.2.0 NVIDIA 418.67
GL_SHADING_LANGUAGE_VERSION=1.50 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.2.0 NVIDIA 418.67
Vendor = NVIDIA Corporation
Renderer = TITAN V/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
Traceback (most recent call last):
  File "examples/demo/play_husky_nonviz.py", line 14, in <module>
    env = HuskyNavigateEnv(config = args.config)
  File "/root/mount/gibson/gibson/envs/husky_env.py", line 38, in __init__
    tracking_camera=tracking_camera)
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 333, in __init__
    self.model_path = get_model_path(self.model_id)
  File "/root/mount/gibson/gibson/data/datasets.py", line 52, in get_model_path
    assert (model_id in os.listdir(data_path)) or model_id == 'stadium', "Model {} does not exist".format(model_id)
AssertionError: Model space7 does not exist
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/miniconda/envs/py35/lib/python3.5/site-packages/gym/utils/closer.py", line 67, in close
    closeable.close()
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 498, in _close
    self.UI._close()
AttributeError: 'HuskyNavigateEnv' object has no attribute 'UI'
Exception ignored in: <bound method Env.__del__ of <gibson.envs.husky_env.HuskyNavigateEnv object at 0x7f3b37df90b8>>
Traceback (most recent call last):
  File "/miniconda/envs/py35/lib/python3.5/site-packages/gym/core.py", line 203, in __del__
    self.close()
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 498, in _close
    self.UI._close()
AttributeError: 'HuskyNavigateEnv' object has no attribute 'UI'
root@b63b6a96e6d2:~/mount/gibson#

Yuantian013, 2019-05-29:
(base) weipan@TUD278688:$ xhost access control enabled, only authorized clients can connect LOCAL: SI:localuser:weipan (base) weipan@TUD278688:$ sudo

Yuantian013, 2019-05-29:
### _**(base) weipan@TUD278688:~$ xhost
access control enabled, only authorized clients can connect
LOCAL:
SI:localuser:weipan
(base) weipan@TUD278688:~$ sudo docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v LOCAL:/root/mount/gibson/gibson/assets/dataset xf1280/gibson:0.3.1
root@b63b6a96e6d2:~/mount/gibson# python examples/demo/play_husky_nonviz.py
pybullet build time: Sep 27 2018 00:17:23
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
/root/mount/gibson/examples/demo/../configs/play/play_husky_nonviz.yaml
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Creating context
Created GL 3.0 context
Direct GLX rendering context obtained
Making context current
GL_VENDOR=NVIDIA Corporation
GL_RENDERER=TITAN V/PCIe/SSE2
GL_VERSION=3.2.0 NVIDIA 418.67
GL_SHADING_LANGUAGE_VERSION=1.50 NVIDIA via Cg compiler
pthread_getconcurrency()=0
Version = 3.2.0 NVIDIA 418.67
Vendor = NVIDIA Corporation
Renderer = TITAN V/PCIe/SSE2
b3Printf: Selected demo: Physics Server
startThreads creating 1 threads.
starting thread 0
started thread 0 
MotionThreadFunc thread started
Traceback (most recent call last):
  File "examples/demo/play_husky_nonviz.py", line 14, in <module>
    env = HuskyNavigateEnv(config = args.config)
  File "/root/mount/gibson/gibson/envs/husky_env.py", line 38, in __init__
    tracking_camera=tracking_camera)
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 333, in __init__
    self.model_path = get_model_path(self.model_id)
  File "/root/mount/gibson/gibson/data/datasets.py", line 52, in get_model_path
    assert (model_id in os.listdir(data_path)) or model_id == 'stadium', "Model {} does not exist".format(model_id)
AssertionError: Model space7 does not exist
numActiveThreads = 0
stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
finished
numActiveThreads = 0
btShutDownExampleBrowser stopping threads
stopThreads: Thread 0 used: 1
Thread with taskId 0 exiting
Thread TERMINATED
destroy semaphore
semaphore destroyed
destroy main semaphore
main semaphore destroyed
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/miniconda/envs/py35/lib/python3.5/site-packages/gym/utils/closer.py", line 67, in close
    closeable.close()
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 498, in _close
    self.UI._close()
AttributeError: 'HuskyNavigateEnv' object has no attribute 'UI'
Exception ignored in: <bound method Env.__del__ of <gibson.envs.husky_env.HuskyNavigateEnv object at 0x7f3b37df90b8>>
Traceback (most recent call last):
  File "/miniconda/envs/py35/lib/python3.5/site-packages/gym/core.py", line 203, in __del__
    self.close()
  File "/root/mount/gibson/gibson/envs/env_modalities.py", line 498, in _close
    self.UI._close()
AttributeError: 'HuskyNavigateEnv' object has no attribute 'UI'
root@b63b6a96e6d2:~/mount/gibson#**_

fxia22, 2019-05-29:
It's because space7 doesn't exist in your dataset, did you download the dataset?

lullabee, 2022-03-18:
Which dataset needs to be downloaded?
Space7 doesn't look to be anywhere in the docker image, nor in ig_dataset

letthink, 2024-03-02:
Hi, firstly you need download the dataset from 
`# download the dataset from https://storage.googleapis.com/gibson_scenes/dataset.tar.gz` 
Secondly, maybe your data path has something wrong, for example: `/home/gibson/dataset` is right, but `/home/gibson/` is wrong.