StanfordVL/GibsonEnv #94: Failed to retrieve a framebuffer config using Docker

aDobeZ, 2019-07-05:
I tried to install Gibson by Docker. The installation went well in both methods, but when I tried to run demos such as `python examples/demo/play_husky_nonviz.py`, the result turned out to be:
`docker run --runtime=nvidia -ti --rm -e DISPLAY -v /tmp/.X11-unix:/tmp/.X11-unix -v ~/GibsonEnv/gibson/assets/dataset:/root/mount/gibson/gibson/assets/dataset gibson
root@7cf38bcc3edd:~/mount/gibson# python examples/demo/play_husky_nonviz.py 
pybullet build time: Jul  5 2019 14:47:50
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
/root/mount/gibson/examples/demo/../configs/play/play_husky_nonviz.yaml
/root/mount/gibson/gibson/envs/env_bases.py:66: YAMLLoadWarning: calling yaml.load() without Loader=... is deprecated, as the default Loader is unsafe. Please read https://msg.pyyaml.org/load for full details.
  config_data = yaml.load(f)
startThreads creating 1 threads.
starting thread 0
started thread 0 
argc=2
argv[0] = --unused
argv[1] = --start_demo_name=Physics Server
ExampleBrowserThreadFunc started
X11 functions dynamically loaded using dlopen/dlsym OK!
Failed to retrieve a framebuffer config
`
I was wondering why. Looking forward to your reply!

fxia22, 2019-07-05:
Did you run `xhost +local:root` ?

aDobeZ, 2019-07-08:
> Did you run `xhost +local:root` ?

Yes, of course. But it's still not working.

aDobeZ, 2019-07-24:
Anyway, I successfully installed it by building from source. Thanks all the same!

closed 2019-07-24 by aDobeZ

whatseven, 2019-09-10:
For people who face the same issues: I fix this issue by reinstalling the Nvidia driver without `--no-opengl-files` flags on ubuntu18.04, which may be considered to solve the Login screen loop question in ubuntu16. But in ubuntu18, I login my computer successfully without this flag.