StanfordVL/GibsonEnv #95: Automatic navigation to target using provided waypoints

ashwinram10, 2019-07-16:
I am using turtlebot_navigation in Gibson ROS. I need to record RGBD observations and ground-truth pose along sample trajectories automatically.

I initially tried to extract maps based on #49 and plan and navigate using Move_base. But the robot get stuck in between as the plans are not exactly correct (inaccurate maps).

Is there any way to automatically control the turtlebot in Gibson and navigate to goal via waypoints provided for each environment ? @fxia22 @hzyjerry 

For eg. Initialize robot at (startx,starty,startz) and navigate to waypoint1. Then navigate from waypoint1 to waypoint2 .... till goal is reached. 

**Sample Path from Allensville:**
```

"dist": 5.799,
        "goalX": 6.198, 
        "goalY": 4.944,
        "goalZ": 0.476,
        "id": "Allensville",
        "level": 1,
        "pathDist": 7.591,
        "split": "full",
        "startAngle": 6.191,
        "startX": 1.049, 
        "startY": 2.277,
        "startZ": 0.468,
        "waypoints": [
            [
                1.049,
                2.277,
                0.468
            ],
            [
                1.848,
                2.878,
                0.467
            ],
            [
                1.922,
                2.933,
                0.466
            ],.....
]
```

Thanks in Advance!!

Berk035, 2019-07-17:
I am looking forward also.

zubair-irshad, 2019-09-25:
Hi @ashwinram10 did you make progress with this issue ? I am also working on the same problem. Let me know if you want to discuss way forward for this issue.