StanfordVL/GibsonEnv #101: Dataset Images of 'Forkland'

Berk035, 2019-11-04:
Hi everyone,

I've tried 'Forkland' as a new model just before. I set initial orientation and position of the agent regularly. Unfortunately, I couldn't get usable rgb or depth images from user interface and programme. I tried also pass different level in order to change floor with config file, it doesn't effect this. Do you have any idea for this issue?

[image]

fxia22, 2019-11-20:
Looks like it is the same as this issue: https://github.com/StanfordVL/GibsonEnv/issues/22
It looks like depth_render opened a wrong model, that might come from the last session where the process didn't terminate properly. You can run ps -ef | grep depth_render to find that process and kill with pkill depth_render then start Gibson again.

Berk035, 2019-11-20:
Thank you for your help Fei. It looks good now.

closed 2019-11-20 by Berk035