StanfordVL/GibsonEnv #102: error when run examples/train/train_husky_navigate_ppo2.py

gophermmq, 2019-11-25:
I got this error when running examples/train/train_husky_navigate_ppo2.py

File "/home/volkan/research/gibson/GibsonEnv/gibson/utils/fuse_policy2.py", line 265, in __init__
 TypeError: fc() got an unexpected keyword argument 'act'

Any thoughts on solving this problem? I can run 'examples/train/train_husky_navigate_ppo2.py' successfully.

Berk035, 2019-11-26:
Hi Volkan,

You said you could able to run examples/train/train_husky_navigate_ppo2.py script. I couldn't understand the problem completely. fc() got an unexpected keyword argument 'act' error is related with fuse_policy2.py while you were trying to initiate variables. It seems to your act variable is not defined in that class.

gophermmq, 2019-12-02:
Hi, 

Thank you very much for your reply. I solved the issue by reinstalling (upgrade the version) of the gym library. Now I could run the script and see some statistics on the terminal without any problem.

Am I supposed to see a window for the camera outputs and robot visual locations? What I see is only a terminal. But there are visual outputs when running 'python examples/demo/play_husky_camera.py'.

gophermmq, 2019-12-02:
OK. I see. I should run 'enjoy_husky_navigate_ppo1/2' to see the results.

closed 2019-12-02 by gophermmq