StanfordVL/GibsonEnv #106: Cannot navigate the husky to render images: Unable to initialize EGL

chaundm, 2019-12-17:
Please help me to fix this issue when I run this command
python examples/demo/play_husky_camera.py
I cannot do anything in GUI environment except zooming the room. The RGB images show me black screen while the GUI environment does not respond

but this command for navigating husky works perfectly 
python examples/demo/play_husky_nonviz.py

And here is the screenshot for my problem

[image]

fxia22, 2019-12-18:
can you show the full log? Thanks

chaundm, 2019-12-19:
Thank you for your respond

Please find my following log file as screenshot. As you can see, it stopped at the line "Episode count: 0"

And here is the details of my pc:
Ubuntu: 18.04
CUDA 10.1
RTX2070 (eGPU): 8Gb
NVIDIA driver: 418

[image]

fxia22, 2019-12-19:
Got it, can you try the methods here and see if it gets resolved? http://svl.stanford.edu/gibson2/docs/issues.html

chaundm, 2019-12-20:
Thanks a lot, I solved it by stopping X server 

[image]

closed 2019-12-20 by chaundm