StanfordVL/GibsonEnv #107: Super slow simulation after adding "rgb_filled' in output

zubair-irshad, 2019-12-23:
Simulations gets super slow after I add "rgb_filled" to my output within the config file. Any leads would be appreciated!

zubair-irshad, 2019-12-23:
[image]

Berk035, 2019-12-24:
Hello @zubair-irshad,

RGB_FILLED may cause that lots of data calculation in simulation as DEPTH mode uses less. Another option for problem is sensitive timestep and frame_skip values. I suggest to you increasing frame_skip values in config file. Have a nice day.

zubair-irshad, 2019-12-24:
Awesome. Thabks @Berk035 
May I ask what timestep and frame_skip values worked for you? Also what fps did you get after optimization?

closed 2023-01-20 by zubair-irshad