StanfordVL/GibsonEnv #109: Problems on running the example of play_huskey_camera.py

HuYue-CS, 2020-01-13:
After installing from the source, I failed to run the the example of play_husky_camera while there occurred an error of FileNotFoundError. At the line 673 of /gibson/envs/env_modalities.py, the program cannot find the file ./depth_render in the path /gibson/core/channels/depth_render. Actually, there is no such a file in the same relative path in the github repository.

Berk035, 2020-01-13:
Hi @HuYue-CS,

You should install depth_render properly. Please fix the errors while compiling codes.

fxia22, 2020-01-13:
@HuYue-CS  You would need to compile the source code with `./build.sh build_local`

HuYue-CS, 2020-01-13:
> @HuYue-CS You would need to compile the source code with `./build.sh build_local`

I have run this script but there were some errors related to the GNU version that is not supported by gcc versions later than 6. But after I installed gcc 5.5.0, it still cannot build with more errors. I guess there are some complicated dependencies so it's better for me to install from your docker image.