StanfordVL/GibsonEnv #115: About adding an obstacle to environment

Berk035, 2020-02-12:
Hello dear contributors,

I wondered whether I could add an obstacle to environment or not in 'pybullet' environment. Also, The obstacle have to be seen by depth camera on the agent which managing its policy in order to avoid obstacle. If you can help, I will be very appreciate. 

Thanks in advance.

fxia22, 2020-02-13:
Hi @Berk035 , great question! For adding objects, please use our latest version of Gibson Environment named interactive Gibson Environment, see project here: http://svl.stanford.edu/gibson2/

Code: https://github.com/StanfordVL/GibsonEnvV2

Berk035, 2020-02-13:
> Hi @Berk035 , great question! For adding objects, please use our latest version of Gibson Environment named interactive Gibson Environment, see project here: http://svl.stanford.edu/gibson2/
> 
> Code: https://github.com/StanfordVL/GibsonEnvV2

Hi @fxia22, It's very nice to hear, I have been waiting this for a long time. I follow your studies with excitement. So, I will work on V1 until my studies are finished. Then, I can study on V2 with pleasure. Thank you for attention.

closed 2020-02-13 by Berk035