StanfordVL/GibsonEnv #127: Can not download dataset from storage.googleapis

seungju-mmc, 2021-05-06:
I run a command in ubuntu terminal
wget https://storage.googleapis.com/gibsonassets/dataset.tar.gz
instead of download, I got the forbidden message

Also, i access the path by chrome.
[image]

leokonto, 2021-05-12:
I am having the same problem and get `Error: 403` when executing `/.downloads.sh`

fxia22, 2021-05-12:
Just fixed it, feel free to try again.

kohjingyu, 2021-06-28:
I'm experiencing this issue again, as of today.

fxia22, 2021-06-29:
@kohjingyu I just tried and things seem to be fine, can you file in the gibson dataset agreement again to get the new links?

kohjingyu, 2021-06-29:
@fxia22 Thanks for the quick response - the new links work!

closed 2021-07-02 by fxia22