StanfordVL/GibsonEnv #129: RGB_FILLED failed but depth successed

rainprob, 2021-07-15:
[image]

rainprob, 2021-07-15:
I tried RGB_PREFILLED, NORMAL, and SEMENTICS. 
RGB_PREFILLED can display the correct content but it still has some wrong pixels, the results of NORMAL and SEMENTICS are both black. _
[image]

rainprob, 2021-07-16:
I tried all solutions in issues, such as checking the version of torchvision and torch, kill depth_render process, and others. But these all don't work.

Berk035, 2021-07-27:
Hello @rainprob ,

I recommend to reinstall the GibsonEnv clearly. You can try to install docker version instead of building source. Have a nice day.

fxia22, 2021-07-28:
@rainprob I observe similar issues when the `gibson/core/render/render_cuda_f.so` is not correctly compiled with nvcc. Can you check your `nvcc --version` is the same as `nvidia-smi` version?

rainprob, 2021-07-29:
@fxia22 
You are right. My nvcc --version is 11.0, however nvidia-smi version is 11.2. So, how can I solve that? Make them both versions same？

rainprob, 2021-07-29:
@Berk035 
Thanks for your reply.

rainprob, 2021-07-29:
> @fxia22
> You are right. My nvcc --version is 11.0, however nvidia-smi version is 11.2. So, how can I solve that? Make them both versions same？

In gibson docker container, nvcc --version is 9.0 and nvdia-smi version is 11.2.

fxia22, 2021-07-29:
It seems that you have multiple versions of cuda installed in `/usr/local`, you can expose the version used in `nvidia-smi` by soft linking `cuda` to it. For example:

[image]

rainprob, 2021-07-30:
@fxia22 
I tried to download cuda11.2 for runtime that has the same version as cuda driver version. But the results shown in web_ui is still the same. 
[image]

fxia22, 2021-07-30:
@rainprob did you build the code again? You would need to run `./build.sh build_local` in GibsonEnv folder.

rainprob, 2021-08-03:
@fxia22 yes, but the buiding result shows some warnings. I am not sure if the warnings mean building failure and will lead a wrong camera output.

rainprob, 2021-08-03:
@fxia22 
[image]