StanfordVL/GibsonEnv #130: Not able to build GibsonEnv locally

adipandas, 2021-07-22:
I am trying to build the GibsonEnv locally. I am using **Ubuntu 20.04**.

I haven't worked with glew/glfw before. And I couldn't figure out the reason behind this error.

I would really appreciate if someone is able to provide feedback/solution to go around this issue.

Please let me know if more information is required.

Thank you.

I am getting the following error after running the ``./build.sh build_local `` script:

```bash
$ ./build.sh build_local 

-- The C compiler identification is GNU 9.3.0
-- The CXX compiler identification is GNU 9.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
CMake Warning (dev) at /usr/share/cmake-3.16/Modules/FindOpenGL.cmake:275 (message):
  Policy CMP0072 is not set: FindOpenGL prefers GLVND by default when
  available.  Run "cmake --help-policy CMP0072" for policy details.  Use the
  cmake_policy command to set the policy and suppress this warning.

  FindOpenGL found both a legacy GL library:

    OPENGL_gl_LIBRARY: /usr/lib/x86_64-linux-gnu/libGL.so

  and GLVND libraries for OpenGL and GLX:

    OPENGL_opengl_LIBRARY: /usr/lib/x86_64-linux-gnu/libOpenGL.so
    OPENGL_glx_LIBRARY: /usr/lib/x86_64-linux-gnu/libGLX.so

  OpenGL_GL_PREFERENCE has not been set to "GLVND" or "LEGACY", so for
  compatibility with CMake 3.10 and below the legacy GL library will be used.
Call Stack (most recent call first):
  CMakeLists.txt:5 (find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

-- Found OpenGL: /usr/lib/x86_64-linux-gnu/libOpenGL.so   
-- Found GLUT: /usr/lib/x86_64-linux-gnu/libglut.so  
-- CMAKE_BINARY_DIR:         /home/adipandas/git_clones/GibsonEnv/gibson/core/channels/build
-- CMAKE_SOURCE_DIR:         /home/adipandas/git_clones/GibsonEnv/gibson/core/channels
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Failed
-- Looking for pthread_create in pthreads
-- Looking for pthread_create in pthreads - not found
-- Looking for pthread_create in pthread
-- Looking for pthread_create in pthread - found
-- Found Threads: TRUE  
-- Found CUDA: /usr/local/cuda-11.4 (found version "11.4") 
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1") 
-- Found GLEW: /usr/include (found version "2.1.0") 
/usr/lib/x86_64-linux-gnu/libGLEW.so
CMake Warning (dev) at /usr/lib/x86_64-linux-gnu/cmake/assimp-5.0/assimpTargets.cmake:54 (if):
  if given arguments:

    "ON"
[... 88 lines cut ...]
      | PFNGLORTHOFPROC
/home/adipandas/git_clones/GibsonEnv/gibson/core/channels/external/glew.c:2779:38: warning: initialization of ‘int’ from ‘void *’ makes integer from pointer without a cast [-Wint-conversion]
 2779 | PFNGLORTHOFOESPROC __glewOrthofOES = NULL;
      |                                      ^~~~
[ 10%] Building C object external/glfw-3.1.2/src/CMakeFiles/glfw.dir/posix_time.c.o
[ 11%] Building C object external/glfw-3.1.2/src/CMakeFiles/glfw.dir/linux_joystick.c.o
[ 12%] Building C object external/glfw-3.1.2/src/CMakeFiles/glfw.dir/posix_tls.c.o
[ 13%] Building C object external/glfw-3.1.2/src/CMakeFiles/glfw.dir/glx_context.c.o
make[2]: *** [external/CMakeFiles/GLEW_1130.dir/build.make:63: external/CMakeFiles/GLEW_1130.dir/glew.c.o] Error 1
/home/adipandas/git_clones/GibsonEnv/gibson/core/channels/external/glfw-3.1.2/src/linux_joystick.c: In function ‘_glfwInitJoysticks’:
/home/adipandas/git_clones/GibsonEnv/gibson/core/channels/external/glfw-3.1.2/src/linux_joystick.c:239:46: warning: ‘%s’ directive output may be truncated writing up to 255 bytes into a region of size 19 [-Wformat-truncation=]
  239 |             snprintf(path, sizeof(path), "%s/%s", dirname, entry->d_name);
      |                                              ^~
/home/adipandas/git_clones/GibsonEnv/gibson/core/channels/external/glfw-3.1.2/src/linux_joystick.c:239:13: note: ‘snprintf’ output 2 or more bytes (assuming 257) into a destination of size 20
  239 |             snprintf(path, sizeof(path), "%s/%s", dirname, entry->d_name);
      |             ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[1]: *** [CMakeFiles/Makefile2:277: external/CMakeFiles/GLEW_1130.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 14%] Linking C static library libglfw3.a
[ 14%] Built target glfw
make: *** [Makefile:130: all] Error 2
/home/adipandas/git_clones/GibsonEnv
/home/adipandas/git_clones/GibsonEnv

```

Berk035, 2021-07-27:
Hello @adipandas ,

I think some compiler libraries of Ubuntu 20.04 are not compatible for Gibson V1. You can also face with problems in OpenGL parts. I recommend to you use Ubuntu 18.04 for Gibson V1.

Sincerely.

adipandas, 2021-07-27:
Hello @Berk035,
I tried it with docker container. It worked out for me.
Thank you.
Best.

closed 2021-07-27 by adipandas

WayneSoong, 2022-01-06:
In Ubuntu 18.04 I met the same problem. It looks like there may be some bug when compiling the glfw.