StanfordVL/GibsonEnv #134: [ERROR] Failed to build mpi4py

adbidwai, 2022-04-30:
System Config:
OS : Ubuntu 18.04

Issue:
The GibsonEnv fails to install at the step : `pip install -e .`
Following is the error log (error at the end):
```
(py35) $USER@Cadence:~/GibsonEnv$ pip install -e .
DEPRECATION: Python 3.5 reached the end of its life on September 13th, 2020. Please upgrade your Python as Python 3.5 is no longer maintained. pip 21.0 will drop support for Python 3.5 in January 2021. pip 21.0 will remove support for this functionality.
Obtaining file:///home/$USER/GibsonEnv
Requirement already satisfied: numpy>=1.10.4 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (1.14.3)
Requirement already satisfied: pyglet>=1.2.0 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (1.5.23)
Requirement already satisfied: gym==0.9.4 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (0.9.4)
Requirement already satisfied: Pillow>=3.3.0 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (5.1.0)
Requirement already satisfied: PyYAML>=3.12 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (3.12)
Collecting pybullet==1.9.4
  Using cached pybullet-1.9.4-cp35-cp35m-linux_x86_64.whl
Collecting transforms3d>=0.3.1
  Using cached transforms3d-0.3.1-py3-none-any.whl
Requirement already satisfied: tqdm>=4 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (4.64.0)
Requirement already satisfied: pyzmq>=16.0.2 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (17.0.0)
Requirement already satisfied: matplotlib>=2.1.0 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (2.2.2)
Collecting mpi4py>=2.0.0
  Using cached mpi4py-3.1.3.tar.gz (2.5 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
    Preparing wheel metadata ... done
Requirement already satisfied: cloudpickle>=0.4.1 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (0.5.3)
Collecting pygame>=1.9.3
  Using cached pygame-2.0.3-cp35-cp35m-manylinux_2_5_x86_64.manylinux1_x86_64.whl (13.3 MB)
Collecting opencv-python
  Using cached opencv_python-4.4.0.42-cp35-cp35m-manylinux2014_x86_64.whl (49.4 MB)
Requirement already satisfied: torchvision==0.2.0 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gibson==0.3.1) (0.2.0)
Collecting aenum
  Using cached aenum-3.1.11-py3-none-any.whl (131 kB)
Requirement already satisfied: requests>=2.0 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gym==0.9.4->gibson==0.3.1) (2.18.4)
Requirement already satisfied: six in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from gym==0.9.4->gibson==0.3.1) (1.11.0)
Requirement already satisfied: torch in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from torchvision==0.2.0->gibson==0.3.1) (0.3.1)
Requirement already satisfied: cycler>=0.10 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from matplotlib>=2.1.0->gibson==0.3.1) (0.10.0)
Requirement already satisfied: pyparsing!=2.0.4,!=2.1.2,!=2.1.6,>=2.0.1 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from matplotlib>=2.1.0->gibson==0.3.1) (2.2.0)
Requirement already satisfied: python-dateutil>=2.1 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from matplotlib>=2.1.0->gibson==0.3.1) (2.7.3)
Requirement already satisfied: pytz in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from matplotlib>=2.1.0->gibson==0.3.1) (2018.4)
Requirement already satisfied: kiwisolver>=1.0.1 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from matplotlib>=2.1.0->gibson==0.3.1) (1.0.1)
Requirement already satisfied: setuptools in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from kiwisolver>=1.0.1->matplotlib>=2.1.0->gibson==0.3.1) (39.1.0)
Requirement already satisfied: chardet<3.1.0,>=3.0.2 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from requests>=2.0->gym==0.9.4->gibson==0.3.1) (3.0.4)
Requirement already satisfied: idna<2.7,>=2.5 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from requests>=2.0->gym==0.9.4->gibson==0.3.1) (2.6)
Requirement already satisfied: urllib3<1.23,>=1.21.1 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from requests>=2.0->gym==0.9.4->gibson==0.3.1) (1.22)
Requirement already satisfied: certifi>=2017.4.17 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from requests>=2.0->gym==0.9.4->gibson==0.3.1) (2018.4.16)
Requirement already satisfied: importlib-resources in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from tqdm>=4->gibson==0.3.1) (3.2.1)
Requirement already satisfied: zipp>=0.4 in /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages (from importlib-resources->tqdm>=4->gibson==0.3.1) (1.2.0)
Building wheels for collected packages: mpi4py
  Building wheel for mpi4py (PEP 517) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/$USER/anaconda3/envs/py35/bin/python /home/$USER/anaconda3/envs/py35/lib/python3.5/site-packages/pip/_vendor/pep517/_in_process.py build_wheel /tmp/tmpr1655c_f
       cwd: /tmp/pip-install-9dzag6bl/mpi4py_1fc915072f08497394730138e604d834
  Complete output (376 lines):
  running bdist_wheel
  running build
  running build_src
  running build_py
  creating build
  creating build/lib.linux-x86_64-3.5
  creating build/lib.linux-x86_64-3.5/mpi4py
  copying src/mpi4py/bench.py -> build/lib.linux-x86_64-3.5/mpi4py
  copying src/mpi4py/run.py -> build/lib.linux-x86_64-3.5/mpi4py
  copying src/mpi4py/__init__.py -> build/lib.linux-x86_64-3.5/mpi4py
  copying src/mpi4py/__main__.py -> build/lib.linux-x86_64-3.5/mpi4py
  creating build/lib.linux-x86_64-3.5/mpi4py/futures
  copying src/mpi4py/futures/_lib.py -> build/lib.linux-x86_64-3.5/mpi4py/futures
  copying src/mpi4py/futures/_base.py -> build/lib.linux-x86_64-3.5/mpi4py/futures
[... 345 lines cut ...]
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `opal_bitmap_t_class'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `opal_hash_table_get_first_key_uint32'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `mca_base_component_list_item_t_class'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `orte_info_register_framework_params'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `mca_base_framework_open'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `orte_rml_recv_cb_t_class'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `orte_session_dir_cleanup'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `opal_info_show_opal_version'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `opal_datatype_add'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `mca_base_var_group_get_count'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `opal_datatype_resize'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `opal_hash_table_set_value_uint64'
  /usr/lib/x86_64-linux-gnu/openmpi/lib/libmpi.so: undefined reference to `opal_list_t_class'
  collect2: error: ld returned 1 exit status
  failure.
  removing: _configtest.c _configtest.o
  error: Cannot link MPI programs. Check your configuration!!!
  ----------------------------------------
  ERROR: Failed building wheel for mpi4py
Failed to build mpi4py
ERROR: Could not build wheels for mpi4py which use PEP 517 and cannot be installed directly

```

Help is appreciated, TIA!!

adbidwai, 2022-05-27:
@fxia22 , @hzyjerry , @amir32002 , @alexsax , @ir0 , @gene-lewis , I am facing this issue since a long time and on multiple systems. It would be really helpful if I could get some advice about this issue. 

Thanks!

Atellas23, 2023-02-21:
Hi! I'm having the same issue. It'd be great if someone could come up with a fix :)

six-finger, 2024-12-12:
remove the ld dir 
"/home/$USER/anaconda3/envs/py35/compiler_compat/ld"

closed 2025-01-15 by adbidwai