StanfordVL/GibsonEnv #136: Meta data raw file is not consistent with what is shown on website

kingsotn, 2022-12-15:
For example, Adrian has 4 floors in .json file but 3 floors on the website

leeked, 2022-12-15:
I think the metadata needs to be updated. The same is true for Albertville and Anaheim.

http://gibsonenv.stanford.edu/models/?id=Albertville
The website says Albertville has 3 floors.
```
{
        "id": "Albertville",
        "name": "model-117",
        "split_full": "train",
        "split_full+": "train",
        "split_medium": "train",
        "split_tiny": "none",
        "stats": {
            "area": 266.125,
            "floor": 4,
            "navigation_complexity": 3.737,
            "room": 16,
            "ssa": 1.297
        }
    },
```

http://gibsonenv.stanford.edu/models/?id=Anaheim
Anaheim should also be 3 floors.
```
{
        "id": "Anaheim",
        "name": "model-479",
        "split_full": "train",
        "split_full+": "train",
        "split_medium": "train",
        "split_tiny": "none",
        "stats": {
            "area": 299.527,
            "floor": 4,
            "navigation_complexity": 2.755,
            "room": 18,
            "ssa": 1.128
        }
    },
```

Based on my observations, I think that the website data is more accurate and should be reflected in the metadata.

chanjuanN, 2024-11-19:
@leeked @kingsotn sorry to bother you! Did you solve this problem? Or **did you know how to know the true and right meta info about these scenes** like Adrian, Anaheim and Albertville, especially the **area** of the scene?
look forward to your reply!!!